EttusResearch/uhd #31: floor rounding issue in gain_group.cpp

guruofquality, 2015-04-02:
Here is small bug with a quick fix: the floor rounding for working around floating point errors in gain_group is incorrect for negative numbers. The consequence is that negative gains will cause the distribution algorithm to be off by a step size. I introduced this bug into the code base years ago. Nevertheless, I would appreciate it if you could upstream this or an equivalent patch into maint. Thanks. 

```
diff --git a/host/lib/utils/gain_group.cpp b/host/lib/utils/gain_group.cpp
index e907a65..0f56eb8 100644
--- a/host/lib/utils/gain_group.cpp
+++ b/host/lib/utils/gain_group.cpp
@@ -51,6 +51,7 @@ static bool compare_by_step_size(
  * \return a multiple of step approximating num
  */
 template <typename T> static T floor_step(T num, T step, T e = T(0.001)){
+    if (num < T(0)) return step*int(num/step - e);
     return step*int(num/step + e);
 }
 
```

bhilburn, 2015-04-22:
Okay, we have replicated this. The patch is going through review and testing, and will go onto maint. Closing the bug to track it internally.

closed 2015-04-22 by bhilburn

guruofquality, 2015-08-15:
**bump** Sorry to bother, but it looks like this patch got lost in the process.

guruofquality, 2016-12-17:
@mbr0wn I know its somewhat of an innocuous bug. It just makes negative numbers round in the wrong direction and being off by 1dB or so isn’t really noticeable. I've been sitting on this patch for a while, and I'm looking for some guidance as to drop it, keep it, ignore it, etc.

mbr0wn, 2017-01-04:
Not sure why it got lost. Reintegrating.