EttusResearch/uhd #34: CentOS 6.4 and 6.5 uhd build failures

sgardner340, 2015-04-14:
The new uhd build fails consistently on my CentOS 6.4 and 6.5 systems. below are the errors that are presented to me when I try to build uhd with my CentOS 6.4 and 6.5 systems.

Note these system were building fine before the last Merge 4 days ago ddae9468b9f69fcbc91c4f8cbffc0b7d69b79b4c

[ 48%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_common.cpp.o  
[ 48%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_version3.cpp.o  
In file included from /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/dboard/db_sbx_common.hpp:  21,:  
 from /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/dboard/db_sbx_common.cpp:  18:   
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In constructor ‘  max2870::max2870(max287x_iface::write_fn)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  266:   error: class ‘  max2870  ’ does not have any field named ‘  max287x  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  266:   error: no matching function for call to ‘  max287x::max287x()  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  228:   note: candidates are: max287x::max287x(max287x_iface::write_fn) [with max287x_regs_t = max2870_regs_t]  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  226:   note: max287x::max287x(const max287x&)  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual double max2870::set_frequency(double, double, double, bool)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  279:   error: ‘  template class max287x  ’ used without template parameters  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual void max2870::commit()  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  285:   error: ‘  template class max287x  ’ used without template parameters  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In constructor ‘  max2871::max2871(max287x_iface::write_fn)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  295:   error: class ‘  max2871  ’ does not have any field named ‘  max287x  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  295:   error: no matching function for call to ‘  max287x::max287x()  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  228:   note: candidates are: max287x::max287x(max287x_iface::write_fn) [with max287x_regs_t = max2871_regs_t]  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  226:  [1;31 In file included from /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/dboard/db_sbx_common.hpp:  21,:  
 from /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/dboard/db_sbx_version3.cpp:  20:   
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In constructor ‘  max2870::max2870(max287x_iface::write_fn)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  266:   error: class ‘  max2870  ’ does not have any field named ‘  max287x  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  266:   error: no matching function for call to ‘  max287x::max287x()  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  228:   note: candidates are: max287x::max287x(max287x_iface::write_fn) [with max287x_regs_t = max2870_regs_t]  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  226:   note: max287x::max287x(const max287x&)  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual double max2870::set_frequency(double, double, double, bool)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  279:   error: ‘  template class max287x  ’ used without template parameters  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual void max2870::commit()  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  285:   error: ‘  template class max287x  ’ used without template parameters  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In constructor ‘  max2871::max2871(max287x_iface::write_fn)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  295:   error: class ‘  max2871  ’ does not have any field named ‘  max287x  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  295:   error: no matching function for call to ‘  max287x::max287x()  ’  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  228:   note: candidates are: max287x::max287x(max287x_iface::write_fn) [with max287x_regs_t = max2871_regs_t]  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  226:   note: max287x::max287x(const max287x&)  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual void max2871::set_muxout_mode(max287x_iface::muxout_mode_t)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  308:   error: ‘  template class max287x  ’ used without template parameters  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual double max2871::set_frequency(double, double, double, bool)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  319:   error: ‘  template class max287x  ’ used without template parameters  
make[2]: *** [lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_version3.cpp.o] Error 1 
make[2]: *** Waiting for unfinished jobs.... 
m note: max287x::max287x(const max287x&)  
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual void max2871::set_muxout_mode(max287x_iface [... cut ...]
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp: In member function ‘  virtual double max2871::set_frequency(double, double, double, bool)  ’: 
/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/rpm-src/src/uhd/host/lib/usrp/common/max287x.hpp:  319:   error: ‘  template class max287x  ’ used without template parameters  
make[2]: *** [lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_common.cpp.o] Error 1 
make[1]: *** [lib/CMakeFiles/uhd.dir/all] Error 2 
make: *** [all] Error 2 
bash return val = 2

mbr0wn, 2015-04-15:
This looks like something caused by compilers unsupported by UHD. If you pull the latest version of UHD maint, does it still break? We added a patch for those old compilers yesterday (00c497b43ffe6d2f0b41163a8620e9ed94780231). Please note that legacy compilers are not officially supported, so at one point in the future they might no longer work with UHD.

mbr0wn, 2015-04-15:
For reference, our list of supported compilers: http://files.ettus.com/manual/page_build_guide.html#cpp_compiler

@sgardner340, can you please provide us feedback if the patch mentioned above fixes your compile, and if not, which compiler version you're using?
Thanks!

sgardner340, 2015-04-15:
My Compilers / Libraries 
GCC = 4.4.7
CMake = 2.8.6
Boost = 1.53
LibUSB = 1.0.9
Python = 2.6

I am running a build now with the 00c497b branch

mbr0wn, 2015-04-15:
Thanks -- please tell is if that fixes the issue. If not, your compiler is on the list of supported ones, and so are your other libs, so we'd want to fix that ASAP.

mbr0wn, 2015-04-15:
BTW, just in case I was unclear, the branch you want to try is either current maint, or the 3.8.3 release candidate.

sgardner340, 2015-04-15:
I am getting a new failure 
We are also having some system problems, I will keep  you posted....
14:15:12 bashexec (/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/src-rpm/src/gr-buffalo/gr-bloodaxe/build):: 
14:15:12     find . -name '*.make' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
14:15:12     find . -name 'Makefile' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
14:15:12     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/cc
14:15:12     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/gcc
14:15:12     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/c++
14:15:12     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/g++
14:15:12     make -j4
14:15:12 
14:15:13 Scanning dependencies of target pygen_python_fba35
14:15:13 Scanning dependencies of target _bloodaxe_swig_swig_tag
14:15:13 Scanning dependencies of target _bloodaxe_swig_doc_tag
14:15:13 [  1%] [  1%] [  1%] Generating __init__.pyc, bloodaxe_gui.pyc
14:15:13 Building CXX object swig/CMakeFiles/_bloodaxe_swig_swig_tag.dir/_bloodaxe_swig_swig_tag.cpp.o
14:15:13 Building CXX object swig/CMakeFiles/_bloodaxe_swig_doc_tag.dir/_bloodaxe_swig_doc_tag.cpp.o
14:15:13 [  2%] Generating __init__.pyo, bloodaxe_gui.pyo
14:15:13 Linking CXX executable _bloodaxe_swig_swig_tag
14:15:13 Linking CXX executable _bloodaxe_swig_doc_tag
14:15:13 [  2%] Built target pygen_python_fba35
14:15:13 Scanning dependencies of target pygen_apps_dfff5
14:15:13 [  3%] Shebangin bloodaxe_eqslicer_example.grc
14:15:13 [  3%] [  3%] Built target _bloodaxe_swig_swig_tag
14:15:13 Built target _bloodaxe_swig_doc_tag
14:15:13 [  4%] Scanning dependencies of target pygen_swig_24b6e
14:15:13 Shebangin bloodaxe_blind_equaliser_example.grc
14:15:13 [  4%] Shebangin bloodaxe_equpdate_example.grc
14:15:13 [  5%] Generating doxygen xml for bloodaxe_swig_doc docs
14:15:13 [  6%] [  6%] Shebangin bloodaxe_no_equaliser_example.grc
14:15:13 Shebangin bloodaxe_demod_block_example.grc
14:15:13 [  7%] Shebangin bloodaxe_demod_block_mt_example.grc
14:15:13 [  8%] Shebangin bloodaxe_timeslicer_example.grc
14:15:13 [  8%] Built target pygen_apps_dfff5
14:15:13 [  9%] Generating bloodaxe_swig_doc.i
14:15:15 Scanning dependencies of target gnuradio-bloodaxe
14:15:15 [  9%] Generating bloodaxe_swig.tag
14:15:15 [ 10%] Swig source
14:15:16 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.4_Desktop_x86_VM/satisfy_order/src-rpm/src/gr-buffalo/gr-bloodaxe/swig/bloodaxe_swig.i:11: Error: Unable to find 'es_queue.i'
14:15:16 make[2]: *** [swig/bloodaxe_swigPYTHON_wrap.cxx] Error 1
14:15:16 make[1]: *** [swig/CMakeFiles/pygen_swig_24b6e.dir/all] Error 2
14:15:16 make[1]: *** Waiting for unfinished jobs....
14:15:16 [ 10%] [ 11%] [ 12%] [ 12%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/bist/bloodaxe_bist_source.cpp.o
14:15:16 Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/bist/bloodaxe_bist_source_impl.cpp.o
14:15:16 Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/resampler/bloodaxe_resampler.cpp.o
14:15:16 Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/resampler/bloodaxe_resampler_impl.cpp.o
14:15:19 [ 13%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/resampler/bloodaxe_resampler_grcp.cpp.o
14:15:19 [ 13%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/resampler/resampler_delegate.cpp.o
14:15:19 [ 14%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/resampler/acquisition_buffer.cpp.o
14:15:19 [ 14%] [ 15%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/equaliser/bloodaxe_equaliser.cpp.o
14:15:19 Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/resampler/batch_size_calculations.cpp.o
14:15:20 [ 16%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/equaliser/bloodaxe_equaliser_impl.cpp.o
14:15:22 [ 17%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/slicer/bloodaxe_slicer.cpp.o
14:15:22 [ 17%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/slicer/bloodaxe_slicer_impl.cpp.o
14:15:23 [ 18%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/eqslicer/bloodaxe_eqslicer.cpp.o
14:15:24 [ 18%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/eqslicer/bloodaxe_eqslicer_impl.cpp.o
14:15:25 [ 19%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/eqslicer/bloodaxe_eqslicer_grcp.cpp.o
14:15:26 [ 20%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/eqslicer/eqslicer_delegate.cpp.o
14:15:27 [ 20%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/eqslicer/bloodaxe_eqslicer_pdu.cpp.o
14:15:28 [ 21%] Building CXX object lib/CMakeFiles/gnuradio-bloodaxe.dir/eqslicer/bloodaxe_eqslicer_pdu_impl.cpp.o
[... cut ...]
14:17:25 Build step 'Execute shell' marked build as failure
14:17:26 [WARNINGS] Skipping publisher since build result is FAILURE
14:17:26 Email was triggered for: Still Failing
14:17:26 Trigger Failure was overridden by another trigger and will not send an email.
14:17:26 Sending email for trigger: Still Failing
14:17:26 Sending email to: sgardner340@gmail.com tim.newman@gmail.com jbk@ltsnet.net
14:17:27 
14:17:27 Deleting project workspace... 
14:17:27 Skipped based on build state FAILURE
14:17:27 
14:17:27 Finished: FAILURE

mbr0wn, 2015-04-15:
@sgardner340, this error log looks like the bug is in a module called gr-bloodaxe, not UHD (gr-bloodaxe is a cool module name, btw).

mbr0wn, 2015-04-16:
@sgardner340, I'll close this unless you still have issues building UHD.

closed 2015-04-16 by mbr0wn

sgardner340, 2015-04-16:
A agree until I get my new issues solved

mbr0wn, 2015-04-16:
Thanks! Just open a new issue, or ask on the mailing list.