EttusResearch/uhd #38: Porting UHD to FreeBSD

rmx-it, 2015-06-28:
Hi,
I'm trying to port and afterwards maintain UHD to FreeBSD.
My primary target is to get UHD working with my B200.
FreeBSD version is 11-current, UHD version is 3.8.4.

Compilation works using the following Patches:
```
--- CMakeLists.txt.orig 2014-12-27 11:05:46.000000000 +0000
+++ CMakeLists.txt      2014-12-27 11:06:18.000000000 +0000
@@ -299,5 +299,5 @@
 UHD_INSTALL(
     FILES ${CMAKE_CURRENT_BINARY_DIR}/uhd.pc
-    DESTINATION ${LIBRARY_DIR}/pkgconfig
+    DESTINATION libdata/pkgconfig
     COMPONENT "devel"
 )
...skipping...
--- examples/CMakeLists.txt.orig        2014-12-27 12:46:00.000000000 +0000
+++ examples/CMakeLists.txt     2014-12-27 12:46:39.000000000 +0000
@@ -51,4 +51,5 @@
     UHD_INSTALL(TARGETS ${example_name} RUNTIME DESTINATION ${PKG_LIB_DIR}/examples COMPONENT examples)
 ENDFOREACH(example_source)
+TARGET_LINK_LIBRARIES(network_relay -pthread)
 
 ########################################################################
...skipping...
--- include/uhd/transport/nirio/nirio_driver_iface.h.orig       2014-12-27 10:53:24.000000000 +0000
+++ include/uhd/transport/nirio/nirio_driver_iface.h    2014-12-27 10:54:03.000000000 +0000
@@ -29,5 +29,5 @@
         #include <WinIoCtl.h>
     #pragma warning(default:4201)
-#elif !defined(UHD_PLATFORM_LINUX)
+#elif !defined(UHD_PLATFORM_LINUX) && !defined(UHD_PLATFORM_BSD)
     #include <IOKit/IOKitLib.h>
 #endif
@@ -78,5 +78,5 @@
 
 //Device handle definition
-#if defined(UHD_PLATFORM_LINUX)
+#if defined(UHD_PLATFORM_LINUX) || defined(UHD_PLATFORM_BSD)
     typedef int rio_dev_handle_t;
 #elif defined(UHD_PLATFORM_WIN32)
```

uhd_find seems to work:
```
(ux32)(root) # uhd_find_devices
FreeBSD 11; Clang version 3.6.1 (tags/RELEASE_361/final 237755); Boost_105500; UHD_003.008.004-0-unknown

-- Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex... done
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    type: b200
    name: 
    serial: F5725B
    product: B200
```

But during uhd_usrp_probe I get:
```
(ux32)(root) # uhd_usrp_probe 
FreeBSD 11; Clang version 3.6.1 (tags/RELEASE_361/final 237755); Boost_105500; UHD_003.008.004-0-unknown

-- Loading FPGA image: /usr/local/share/uhd/images/usrp_b200_fpga.bin... done
-- Operating over USB 3.
-- Detecting internal GPSDO.... No GPSDO found
Error: AssertionError: accum_timeout < _timeout
  in boost::uint64_t radio_ctrl_core_3000_impl::wait_for_ack(const bool)
  at /usr/ports/comms/uhd/work/uhd-3.8.4/lib/usrp/cores/radio_ctrl_core_3000.cpp:230
```

Subsequent calls of uhd_usrp_probe produce:
```
(ux32)(root) # uhd_usrp_probe
FreeBSD 11; Clang version 3.6.1 (tags/RELEASE_361/final 237755); Boost_105500; UHD_003.008.004-0-unknown

-- Operating over USB 3.

UHD Error:
    An unexpected exception was caught in a task loop.
    The task loop will now exit, things may not work.
    RuntimeError: usb rx8 transfer status: 1
Error: AssertionError: accum_timeout < _timeout
  in boost::uint64_t radio_ctrl_core_3000_impl::wait_for_ack(const bool)
  at /usr/ports/comms/uhd/work/uhd-3.8.4/lib/usrp/cores/radio_ctrl_core_3000.cpp:230
```

Now I would need some help to debug this issue further.
Could it be a failure of the firmware transfer?
Is this a known error?

Thanks in advance!

Greetings
---
Michael

mbr0wn, 2015-06-29:
Hey @rmx-it, interesting stuff! We have seen those errors, typically with bad USB controllers, though. Can you confirm the same hardware works with a different OS? What about USB2?

This looks like a bit of a discussion, so I'd encourage you to take this to the usrp-users mailing list.

rmx-it, 2015-07-01:
Hi,
yes its a Asus Zenbook UX32VD-R4002V (Intel I7 + Intel chipset).
It's a USB3.0 only notebook so I couldn't test USB2.0 so far on this puppy.
Other successfully tested OS: Archlinux (Worked best so far).
Your liveusb distribution.

The tricky thing on Linux is the RF-Bandwidth though.
It seems that if any Battery-saving is active (like on most Linux distributions nowadays)
I get rather limited Bandwidth over USB3.0.
This as a tip for your coming liveusb distribution: 
  Set the System into performance mode at full CPU speed without freq-throtteling.
But thats a different issue...

For now I would like to bring FreeBSD on-par with the other OS's.

Thanks for your efforts!

---
Michael

mbr0wn, 2015-07-01:
@rmx-it We do recommend turning off powersave governors in our manual. Since this seems to address mostly FreeBSD-specific issues, I'm going to close this issue. Again, I do invite you to discuss this on the mailing list!

rmx-it, 2015-07-05:
Hi,
sorry for not being clear enough (and the delay of course).

The issue with powersave govenors occured only on Linux for me and has nothing to do with my issue on FreeBSD!

So the original question remains: How to debug this further for the FreeBSD port?

Or more precisely:
- Is there a DEBUG-mode for UHD/the FW-loader I could switch on during compile/run-time?
- How can I assure that the firmware for the USB-controller and the FPGA is properly transferred?

Thanks!
---
Michael

mbr0wn, 2015-07-06:
@rmx-it, it does seem like this is a general support discussion as opposed to a specific bug. Let's move this discussion to the mailing list. I'll be closing this bug.

closed 2015-07-06 by mbr0wn