EttusResearch/uhd #39: CentOS

sgardner340, 2015-07-17:
Hi

I was  building gnuradio on Centos 6.4, 6.5, and 6.6. The version of gnuradio that was being built is:

gnuradio    55d8f482f5acf33b8f629556a4fa5019908bb4be git://http://www.gnuradio.org/git/gnuradio.git

UHD version is :
uhd   012381d999c4a895593412aaf06e73432b458810 git://https://github.com/EttusResearch/uhd.git

This was building fine until the 07/14/2015 when I started getting the following error on all of my CentOS systems:

09:26:24 -- Using install prefix: /apps/smokey64
09:26:24 -- Building for version: v3.7.7.1-204-g55d8f482 / 3.7.8git
09:26:24 CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
09:26:24 Please set them or make sure they are set and tested correctly in the CMake files:
09:26:24 UHD_LIBRARIES
09:26:24     linked by target "gnuradio-uhd" in directory /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/src/gnuradio/gr-uhd/lib
09:26:24     linked by target "_uhd_swig" in directory /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/src/gnuradio/gr-uhd/swig
09:26:24 
09:26:24 -- Configuring incomplete, errors occurred!
09:26:24 See also "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
09:26:24 See also "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/src/gnuradio/build/CMakeFiles/CMakeError.log".
09:26:24 Current step: (gnuradio :: make)
09:26:24 make
09:26:24 bashexec (/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/src/gnuradio/build):: 
09:26:24     find . -name '*.make' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
09:26:24     find . -name 'Makefile' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
09:26:24     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/cc
09:26:24     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/gcc
09:26:24     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/c++
09:26:24     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.6_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/g++
09:26:24     make -j4
09:26:24 
09:26:24 sed: no input files
09:26:24 sed: no input files
09:26:24 make: *** No targets specified and no makefile found.  Stop.

mbr0wn, 2015-07-17:
@sgardner340, I'm not sure if the issue is UHD, GNU Radio or PyBOMBS. I think a good place to ask about this is the GNU Radio mailing list (discuss-gnuradio).

sgardner340, 2015-07-17:
The current version of gnuradio was pushed on the 07/12/2015 55d8f482f5acf33b8f629556a4fa5019908bb4be and it built on all my CentOS system. The pybombs I am using has not changed since  07/09/2015. I last good CentOS build was 07/14/2015...

mbr0wn, 2015-07-20:
I'm closing this for now -- this doesn't point directly to an issue in UHD. @sgardner340, can you please ask about this on the mailing lists.

closed 2015-07-20 by mbr0wn