EttusResearch/uhd #40: Failure to build LibUHD due to missing dependencies should result in an error

douggeiger, 2015-08-03:
I believe it is an error that LibUHD will *not* be built (e.g. due to some dependency not being available). Unfortunately if LibUHD is not built (e.g. by 'pybombs install uhd') - it is currently not readily apparent (due to the large amount of output, and apparent successful completion after installing the manual/man pages/Doxygen pages). Due to the change in the use of Mako instead of Cheetah, this can cause a great deal of confusion on machines that had previously, successfully installed UHD, and it is not currently detected until e.g. building GNURadio and gr-uhd fails to get built (since LibUHD is not available).

Example shown below. Expected output would be a failure (preferably at cmake-time, highlighting the missing dependency), actual output pasted:

```
$ cmake -DENABLE_LIBUHD=On -Wno-dev ~/Code/uhd/host && make && make install DESTDIR=~/install/testing/
-- The CXX compiler identification is GNU 4.9.2
-- The C compiler identification is GNU 4.9.2
-- Check for working CXX compiler: /usr/lib/ccache/c++
-- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/cc
-- Check for working C compiler: /usr/lib/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- 
-- Configuring the python interpreter...
-- Found PythonInterp: /usr/bin/python (found version "2.7.9") 
-- Python interpreter: /usr/bin/python
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Using UHD Images Directory: OFF
-- Build type not specified: defaulting to release.
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN - Success
-- 
-- Configuring Boost C++ Libraries...
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   date_time
--   filesystem
--   program_options
--   regex
--   system
--   thread
--   unit_test_framework
--   serialization
-- Boost include directories: /usr/include
-- Boost library directories: /usr/lib/x86_64-linux-gnu
-- Boost libraries: /usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libpthread.so
-- Looking for log2
-- Looking for log2 - found
-- 
-- Python checking for Python version 2.6 or greater
-- Python checking for Python version 2.6 or greater - found
-- 
-- Python checking for Mako templates 0.4 or greater
-- Python checking for Mako templates 0.4 or greater - "import mako" failed
-- 
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
--   Disabling LibUHD support.
--   Override with -DENABLE_LIBUHD=ON/OFF
-- 
-- Configuring Examples support...
--   Dependency ENABLE_LIBUHD = OFF
--   Disabling Examples support.
--   Override with -DENABLE_EXAMPLES=ON/OFF
-- 
-- Configuring Utils support...
--   Dependency ENABLE_LIBUHD = OFF
--   Disabling Utils support.
--   Override with -DENABLE_UTILS=ON/OFF
-- 
-- Configuring Tests support...
--   Dependency ENABLE_LIBUHD = OFF
[... 1318 lines cut ...]
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/usrp/multi_usrp.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/usrp_clock/octoclock_eeprom.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/usrp_clock/multi_usrp_clock.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/algorithm.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/assert_has.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/assert_has.ipp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/atomic.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/byteswap.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/byteswap.ipp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/cast.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/csv.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/gain_group.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/log.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/math.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/msg.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/msg_task.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/paths.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/pimpl.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/platform.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/safe_call.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/safe_main.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/static.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/tasks.hpp
-- Up-to-date: /home/douggeiger/install/testing/usr/local/include/uhd/utils/thread_priority.hpp
```

balister, 2015-08-03:
I did this in the OE recipe, just in case I ever mess up and do not build mako before uhd again.

```
-EXTRA_OECMAKE = "-DENABLE_E100=TRUE -DENABLE_E300=TRUE"
+EXTRA_OECMAKE = "-DENABLE_E100=TRUE -DENABLE_E300=TRUE -DENABLE_UHD=TRUE"
```

But yes, I have shared your rage over this problem :)

mbr0wn, 2015-08-03:
@douggeiger we have an issue for this on our internal tracker, and will be fiddling with CMake to make `ENABLE_UHD=On` the default. This means people can still disable libuhd, and only build the manual for example, but if they don't, building will fail in the way you describe above.

Until then, or with older releases, setting `ENABLE_UHD=On` explicitly is the way to go.

Why is it the way it is now? I don't know, it's probably a CMake default. No one ever told CMake that `libuhd` is a more special target than the others.

douggeiger, 2015-08-04:
@mbr0wn - passing in -DENABLE_UHD=ON gives me the same output (note, I used -DENABLE_LIBUHD=On, which according to my reading of the cmake-output is what I should be using). I certainly see the use-case for not wanting to build LibUHD (i.e. documentation-only builds), but what I was hoping was that by passing in an explicit cmake option, we could error-out early when dependencies aren't satisfied properly, and this option could then be added to the pybombs recipe (so that, e.g. 'pybombs install uhd' errors out properly)

mbr0wn, 2015-08-04:
@douggeiger `ENABLE_LIBUHD=On` is correct, mine was a typo. If that won't force an error, that's really a bug, and we will fix it. I'll say it's not super-high-priority, since this behaviour has been around since the inception of UHD and so far it's worked fine, but we will handle it. I'll close this issue for now.

closed 2015-08-04 by mbr0wn

robellebor, 2018-02-08:
where do i enable the ENABLE_LIBUHD=On ??

sgardner340, 2018-02-08:
Sorry I am not on the project any more

szakacsm, 2019-12-03:
@mbr0wn where can be set ENABLE_LIBUHD=On? which file?

mbr0wn, 2019-12-03:
Pass it to CMake (-DENABLE_LIBUHD=ON).

szakacsm, 2019-12-05:
how it works with pybombs?