EttusResearch/uhd #42: Please help me, uhd_find_devices and uhd_usrp_probe error

AprilianiHutagaol, 2015-08-15:
uhd_find_devices 
linux; GNU C++ version 4.6.3; Boost_104900; UHD_003.004.002-0-unknown

uhd_find_devices: symbol lookup error: uhd_find_devices: undefined symbol: _ZN3uhd6device4findERKNS_13device_addr_tENS0_15device_filter_tE

dkozel, 2015-08-15:
Hello April,

The UHD version 3.4.2 is over three years old and there are a considerable number of changes and improvements which have been done since then. I'd strongly suggest updating to the latest, 3.8.5. The build_gnuradio script provides an easy way to install the new version. Having more than one version installed at once is rarely useful and very often troublesome so it is recommended to uninstall the old version before installing the new one.

Here is a description of the script.
https://gnuradio.org/redmine/projects/gnuradio/wiki/InstallingGRFromSource#Using-the-build-gnuradio-script

The following commands would ensure that you have the prerequisites installed and then build the release version of UHD.
```
./build_gnuradio prereqs
./build_gnuradio -ut release_003_008_005 uhd_build
```

I recommend subscribing to the USRP-users mailing list and asking questions there in the future as the GitHub issues are not the best place for support type questions.
http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com

If you have a problem updating send an email to that list and I'll be happy to help you further.

closed 2015-08-16 by mbr0wn

ciimaicela, 2017-09-20:
Hello 
I have the same problem. How do I uninstall the old version?

mbr0wn, 2017-09-26:
That depends on how you installed it. Maybe `apt remove`?

ciimaicela, 2017-09-26:
Thanks but already solved it

oladimeji-abdul, 2019-08-05:
how did you solve it please