EttusResearch/uhd #43: find/print_utility() error on official ubuntu packages

guruofquality, 2015-08-15:
The official ubuntu packages use PKG_LIB_DIR="/usr/lib/uhd" to install the examples and utils.
```
override_dh_auto_configure:
	dh_auto_configure -- -DLIB_SUFFIX="/$(DEB_HOST_MULTIARCH)" -DPKG_LIB_DIR="/usr/lib/uhd"
```

The problem is that the uhd::find_utility() call uses the UHD_LIB_DIR regardless, so the prints end up being incorrect. This is a small patch to use PKG_LIB_DIR in find_utility:

```
diff --git a/host/lib/utils/CMakeLists.txt b/host/lib/utils/CMakeLists.txt
index 369920a..7940350 100644
--- a/host/lib/utils/CMakeLists.txt
+++ b/host/lib/utils/CMakeLists.txt
@@ -123,7 +123,7 @@ SET_SOURCE_FILES_PROPERTIES(
     ${CMAKE_CURRENT_SOURCE_DIR}/paths.cpp
     ${CMAKE_CURRENT_SOURCE_DIR}/images.cpp
     PROPERTIES COMPILE_DEFINITIONS
-    "UHD_PKG_PATH=\"${UHD_PKG_PATH}\";UHD_LIB_DIR=\"lib${LIB_SUFFIX}\""
+    "UHD_PKG_PATH=\"${UHD_PKG_PATH}\";UHD_LIB_DIR=\"lib${LIB_SUFFIX}\";PKG_LIB_DIR=\"${PKG_LIB_DIR}\""
 )
 
 ########################################################################
diff --git a/host/lib/utils/paths.cpp b/host/lib/utils/paths.cpp
index f29318d..ac30d9b 100644
--- a/host/lib/utils/paths.cpp
+++ b/host/lib/utils/paths.cpp
@@ -363,7 +363,7 @@ std::string uhd::find_image_path(const std::string &image_name, const std::strin
 }
 
 std::string uhd::find_utility(std::string name) {
-    return fs::path(fs::path(uhd::get_pkg_path()) / UHD_LIB_DIR / "uhd" / "utils" / name)
+    return fs::path(fs::path(uhd::get_pkg_path()) / PKG_LIB_DIR / "utils" / name)
         .string();
 }
 
```

closed 2015-10-09 by bhilburn

guruofquality, 2016-05-15:
@bhilburn you closed the bug, but it still seems to be a problem with the official ppa:

**its up to date**
```
sudo apt-get install uhd-host
Reading package lists... Done
Building dependency tree       
Reading state information... Done
uhd-host is already the newest version (3.9.4-0ubuntu1~xenial1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

**look at the error message**

```
uhd_find_devices 
linux; GNU C++ version 5.3.1 20160413; Boost_105800; UHD_003.009.004-release

UHD Warning:
    EnvironmentError: IOError: Could not find path for image: usrp_b200_fw.hex
    Using images directory: <no images directory located>
    Set the environment variable 'UHD_IMAGES_DIR' appropriately or follow the below instructions to download the images package.
    Please run:
     "/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"
No UHD Devices Found
```

**but this file doesnt exist**

```
 ls /usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py
ls: cannot access '/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py': No such file or directory
```

**because its here**

```
ls /usr/lib/uhd/utils/uhd_images_downloader.py
/usr/lib/uhd/utils/uhd_images_downloader.py

```

pascal-gujer, 2018-07-26:
> Set the environment variable 'UHD_IMAGES_DIR' appropriately

This is the key... After installing my UHD from source as described here https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux I set the 'UHD_IMAGES_DIR' environment variable appropriately in my .bashrc :
`export UHD_IMAGES_DIR=/usr/local/share/uhd/images`

And now everything is working like a charm. :1st_place_medal:

oladimeji-abdul, 2019-08-06:
which command did you run please

alphafox02, 2023-01-30:
Just wanted to drop in hand say, that incorrect print statement mentioned above still happens in uhd 4.1.0.5 in 2023, when installed from source using /usr prefix.

leandroebner, 2024-03-10:
same here. Not user friendly at all!

DanaGoyette, 2025-08-19:
I installed the software from the package in Ubuntu, and the path is incorrect there (arguably more incorrect, since it isn't even under /usr/lib):

The actual path: /usr/libexec/uhd/utils/uhd_images_downloader.py
There's also a copy of the script: /usr/bin/uhd_images_downloader

It would be nice for it to have mentioned either of the correct locations.