EttusResearch/uhd #44: Expose property for master clock range

CodaCarlson, 2015-09-07:
I would like for the master clock range to be exposed in the property tree. I see auto_tick_rate and tick_rate as well as the dsp rate range, however, at least for the b210 max out at 16Msps without a master_clock_rate arg. I would like to be able to dynamically tell what the possible clock range. I look in the b200_io_impl and it looks like it's just a define. I could just include that define, however, I would rather something be exposed in either multi_usrp or the property tree since I am working with different USRP types.

mbr0wn, 2015-09-29:
This is a reasonable request, and I like the idea. There's a couple of issues that need resolving; for example, the X-Series allows certain discrete master clock rates (albeit not at runtime). The B200- and E310-series already have different ranges, so this would be quite useful there.

mbr0wn, 2018-03-28:
@CodaCarlson We have an actual fix in the pipeline. It's going through review and CI and will probably go live in a few days.

closed 2018-03-28 by mbr0wn