EttusResearch/uhd #45: Pybombs mako problem

jamisxom, 2015-09-16:
Hi, I've been unable to use pybombs to install UHD or anything that depends on UHD.
The issue is possibly something to do with Mako, although I've tried after installing mako using both pybombs and apt-get and still have the same problem.
Pybombs has been able to install other recipies that don't seem to depend on UHD. I've found a few posts about similar problems but most seem to have been solved by installing mako, which hasn't worked for me.
Does anyone have any suggestions?

-- 
-- Python checking for Python version 2.6 or greater
-- Python checking for Python version 2.6 or greater - found
-- 
-- Python checking for Mako templates 0.4 or greater
-- Python checking for Mako templates 0.4 or greater - "import mako" failed
-- 
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
CMake Error at cmake/Modules/UHDComponent.cmake:51 (MESSAGE):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:279 (LIBUHD_REGISTER_COMPONENT)

mbr0wn, 2015-09-16:
@jamisxom, this doesn't look like a UHD bug, rather a path setup problem. I recommend you take this to the mailing list, so we can get a conversation going to fix your problem. In the meantime, I'll be closing this issue.

mbr0wn, 2015-09-16:
When you do contact the mailing list, show us the output of

    $ python -c 'import mako'

closehaulcom, 2015-10-02:
I encountered the same issue.  I attempted to re-install python-mako with the usual sudo apt-get install, command, to no avail.  However, it magically worked after I tried Plan B:

_$ pip install mako_

mbr0wn, 2018-06-22:
It should yes. Which Ubuntu? Which Pybombs?

ndandanov, 2018-06-23:
I just encountered the same issue on a fresh Ubuntu 18.04 x64 install while trying to install `gnuradio` using the [official PyBOMBS recipe](https://github.com/gnuradio/pybombs).
`pip install mako` and `sudo apt-get install python-mako` seems to have resolved it. I will report later if any other issues occur.

mbr0wn, 2018-06-25:
@ndandanov Which PyBOMBS version did you use? Did you install via pip `$ pip install pybombs` or from source?

ndandanov, 2018-06-26:
@mbr0wn I installed it via `pip`. Then I read that this issue has been fixed and installed from source.

mbr0wn, 2018-06-26:
Did that fix it?

ndandanov, 2018-06-26:
Sorry, I think I switched to the latest source for another bug which popped up later during the installation.
I can try to replicate the issue in a fresh Ubuntu install however :) That would be no problem for me.

mbr0wn, 2018-06-28:
No that's fine! It seems to work with the latest pip though.

hz5612777, 2019-06-05:
I have encountered the same problem, even if **mako** is properly installed, it is still useless.  I found it is a version issue. UHD called the python3 version, so pip should correspond to python3.  When I use **pip3 install mako**, the problem is solved.

jarvisAUT, 2019-08-21:
> 
> 
> I have encountered the same problem, even if **mako** is properly installed, it is still useless. I found it is a version issue. UHD called the python3 version, so pip should correspond to python3. When I use **pip3 install mako**, the problem is solved.

Solved it for me.

cbdonohue, 2019-09-04:
Should Pybombs be updated to avoid this?

mbr0wn, 2019-09-22:
PyBOMBS has been updated on master branch to do this.

closed 2019-09-22 by mbr0wn

ahmad-sho, 2021-12-10:
Hi, I had a simmilar problem trying to install gnuradio: I have the following messages:
...
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - not found

I tried to install mako after seeing the suggestions above, but I already have a new version:
Requirement already satisfied: mako in /usr/lib/python3/dist-packages (1.1.0)
any suggestions?
I'm running Ubuntu 20.04.3 with python3 installed.

ahmad-sho, 2021-12-10:
Hi, I had a similar problem trying to install gnuradio: I have the following messages:

-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - not found

I tried to install mako after seeing the suggestions above, but I already have a new version:
Requirement already satisfied: mako in /usr/lib/python3/dist-packages (1.1.0)
any suggestions?
I'm running Ubuntu 20.04.3 with python3 installed.