EttusResearch/uhd #46: uhd doesn't pick up six correctly?

ckuethe, 2015-10-02:
Perhaps a copy of six should be included in UHD?

ckuethe:/home/gnuradio$ uhd_images_downloader 
Traceback (most recent call last):
  File "/home/gnuradio/bin/uhd_images_downloader", line 26, in <module>
    import requests
  File "/usr/lib/python2.7/dist-packages/requests/__init__.py", line 55, in <module>
    import urllib3
  File "/usr/lib/python2.7/dist-packages/urllib3/__init__.py", line 10, in <module>
    from .connectionpool import (
  File "/usr/lib/python2.7/dist-packages/urllib3/connectionpool.py", line 30, in <module>
    import six
ImportError: No module named six
ckuethe:/home/gnuradio$ find /usr/lib -name six\*
/usr/lib/pymodules/python2.7/pymc/six.pyc
/usr/lib/pymodules/python2.7/pymc/six.py
/usr/lib/python3/dist-packages/six-1.9.0.egg-info
/usr/lib/python3/dist-packages/__pycache__/six.cpython-34.pyc
/usr/lib/python3/dist-packages/six.py
/usr/lib/python2.7/dist-packages/astropy/extern/six.pyc
/usr/lib/python2.7/dist-packages/astropy/extern/six.py
/usr/lib/python2.7/dist-packages/astropy/extern/bundled/six.pyc
/usr/lib/python2.7/dist-packages/astropy/extern/bundled/six.py
/usr/lib/python2.7/dist-packages/sklearn/externals/six.pyc
/usr/lib/python2.7/dist-packages/sklearn/externals/six.py
/usr/lib/python2.7/dist-packages/ptrace/six.pyc
/usr/lib/python2.7/dist-packages/ptrace/six.py
/usr/lib/python2.7/dist-packages/zmq/utils/sixcerpt.py
/usr/lib/python2.7/dist-packages/zmq/utils/sixcerpt.pyc
/usr/lib/python2.7/dist-packages/django/utils/six.pyc
/usr/lib/python2.7/dist-packages/django/utils/six.py
/usr/lib/python2.7/dist-packages/scipy/lib/six.pyc
/usr/lib/python2.7/dist-packages/scipy/lib/six.py
/usr/lib/x86_64-linux-gnu/ImageMagick-6.8.9/modules-Q16/coders/sixel.so
/usr/lib/x86_64-linux-gnu/ImageMagick-6.8.9/modules-Q16/coders/sixel.la
ckuethe@ambiflex:/home/gnuradio$ sudo -H pip install six
Downloading/unpacking six
  Downloading six-1.9.0-py2.py3-none-any.whl
Installing collected packages: six
Successfully installed six
Cleaning up...
ckuethe:/home/gnuradio$ uhd_images_downloader 
Images destination:      /home/gnuradio/share/uhd/images
Downloading images from: http://files.ettus.com/binaries/images/uhd-images_003.009.001-release.zip
Downloading images to:   /tmp/tmpHL797j/uhd-images_003.009.001-release.zip
24703 kB / 24703 kB (100%)

Images successfully installed to: /home/gnuradio/share/uhd/images

ckuethe, 2015-10-02:
This is happening on xubuntu 15.04, UHD from git via PyBOMBS

mbr0wn, 2015-10-03:
`six` should be installed if requests is properly installed. It seems like on your system, six was even installed. I'd say this looks like an issue with the Python package management.

closed 2015-10-03 by mbr0wn