EttusResearch/uhd #58: USRP B210: streaming from B210 to disk doesn't work reliably

ptrkrysik, 2016-04-28:
There is problem with overflows when streaming to disk from USRP B210. The problem persists no matter what other hardware is being used. Multiple tests on computers with powerful hardware (with top intel i7 processors, up to 32GB RAM, SSD disks with write throughput over 400MB/s) shown that even for low rates like 5MS/s and recorded period of about 10 minutes there is almost 100% chance that overflow error will appear. With higher rates about 1 minute of recording is enough to get an overflow error.

The problem also doesn't depend on particular application used for the test. It appears when any application, software suit or operating system is being used. Rate of the overflows might be different but it is impossible to avoid them completely. 

An example application that is distributed with the driver can be used to show the nature of this problem:

rx_samples_to_file --rate 10000000 --duration 600

(for streaming with rate of 10MS/s)
The above example works correctly in the following conditions:
-X or N series device is being used as the receiver,
-when streaming from B210 to ramdisk (in this case 40MS/s is easily achievable).

When streaming from B210 to a HDD (SSD or any other type of SATA disk) the problem with overflows always appear.

The conclusion is that it is not fault of the other hardware like disk (it works well with USRP N210 or X310) or USB 3.0 controller (it works when streaming to ramdisk) alone. It might be fault for example of:
-B210 firmware/FPGA code,
-host side of USRP B210 driver,
-competition for resources between USB 3.0 and some other hardware that is being used when streaming to SATA disk.

The problem can be reproduced on any computer with use of the description above. In case you think this is a false statement, please show an example of hardware/software configuration for which the above command was executed at least twice in row without an overflow.

I'm willing to help with the solution of this problem. At this moment I need guidance how to begin with looking for its source and what tools can be used to confirm/reject theories where it might be.

mbr0wn, 2016-04-28:
@ptrkrysik, thanks for the detailed report. We'll be working on B210 improvements in the near future and we'll take this into account.

@dkozel please note.

jfsir, 2016-04-29:
I had this issue for quite a while.  If you set num_recv_frames=1024 in the device args, the issue will go away.

ptrkrysik, 2016-04-30:
@jfsir , can you confirm that this command:
${path to uhd examples}/rx_samples_to_file --rate 10000000 --duration 600 --args="num_recv_frames=1024"
works for you (or any other program - just tell which one and how you invoke it)?

If yes - can you tell what hardware do you use, what distribution and what version of UHD you have installed?

I tried the test above and it doesn't work on my setup - I had an overflow in under one minute. 
Hardware: Dell Latitude E6330 with i7-3540M processor, 16GB RAM, additional SSD disk Samsung 850 EVO 1TB, 
Software: operating system Ubuntu 14.04, kernel 3.19.0-58-generic, UHD 003.010.git-0-42a3eeb6

mbr0wn, 2016-05-02:
I'm closing this for now -- there's still stuff to do, but this issue has turned a bit into a support request. @ptrkrysik, we'll keep you updated on our progress.

closed 2016-05-02 by mbr0wn

ptrkrysik, 2016-05-02:
I strongly disagree. Streaming data to disk is one of basic use cases for SDR's. And nobody here presented a way to show that it actually works for B210 with ANY software/computer. 

I'm not asking you to tell me how to make it work on my computer. But if you're unable to present single case when streaming to disk work reliably with B210 maybe you should warn about that in the device's datasheet.

mbr0wn, 2016-05-02:
@ptrkrysik what I meant is, we need to move this discussion elsewhere, e.g. the mailing list.

CaiXuesong, 2016-06-16:
@ptrkrysik I'm just struggling with the same problem.

dkozel, 2016-06-16:
@CaiXuesong Can you email either support@ettus.com or the usrp-users mailing list please? We can help you there.
http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com

jonk4m, 2021-07-19:
@mbr0wn I am struggling with the same issue.