EttusResearch/uhd #61: sigsegv at convert_common.hpp:156

enesunal, 2016-06-01:
I'm using the library with srsUE (github/srsLTE/srsUE). It's giving SIGSEGV sometimes. Stack trace is below. Hope it will help to see the issue.

#0  item32_sc16_to_xx<&uhd::htowx<unsigned int>, float> (scale_factor=3,0518502025191039e-05, nsamps=<optimized out>, output=0x7f0c5825fd48, input=0x7f0c582d7430)
    at ~/tmp/uhd/host/lib/convert/convert_common.hpp:156
#1  __convert_sc16_item32_le_1_fc32_1_PRIORITY_SIMD::operator() (this=0x24e55f0, inputs=..., outputs=..., nsamps=351) at ~/tmp/uhd/host/lib/convert/sse2_sc16_to_fc32.cpp:62
#2  0x00007f0c6e5a8ae4 in uhd::convert::converter::conv (num=351, out=..., in=..., this=0x24e55f0) at ~/tmp/uhd/host/include/uhd/convert.hpp:44
#3  uhd::transport::sph::recv_packet_handler::converter_thread_task (index=<optimized out>, this=0x2505d88)
    at ~/tmp/uhd/host/lib/usrp/usrp2/../../transport/super_recv_packet_handler.hpp:702
#4  uhd::transport::sph::recv_packet_handler::recv_one_packet (buffer_offset_bytes=15624, timeout=5, metadata=..., nsamps_per_buff=351, buffs=..., this=0x2505d88)
    at ~/tmp/uhd/host/lib/usrp/usrp2/../../transport/super_recv_packet_handler.hpp:666
#5  uhd::transport::sph::recv_packet_handler::recv (one_packet=false, timeout=5, metadata=..., nsamps_per_buff=2304, buffs=..., this=0x2505d88)
    at ~/tmp/uhd/host/lib/usrp/usrp2/../../transport/super_recv_packet_handler.hpp:258
#6  uhd::transport::sph::recv_packet_streamer::recv (this=0x2505d80, buffs=..., nsamps_per_buff=2304, metadata=..., timeout=5, one_packet=false)
    at ~/tmp/uhd/host/lib/usrp/usrp2/../../transport/super_recv_packet_handler.hpp:805
#7  0x00007f0c6e2a10d5 in uhd_rx_streamer_recv (h=0x247c440, buffs=0x7f0c527fbcb0, samps_per_buff=2304, md=0x247a0d0, timeout=5, one_packet=false, items_recvd=0x7f0c527fbca8)
    at ~/tmp/uhd/host/lib/usrp/usrp_c.cpp:140

mbr0wn, 2016-06-03:
@enesunal Can you give us some info about your platform?

enesunal, 2016-06-03:
Sorry, the issue wasn't about uhd.

closed 2016-06-03 by enesunal

mbr0wn, 2016-06-03:
OK, thanks!