EttusResearch/uhd #62: Need support in configuring TX1 in B200/B210

mayuri2016, 2016-06-15:
Default configuration in B200 code is always TX2 as FE1=1. I tried changing this to TX1 making FE1=0 and FE2=1. But this doesn't help. I don't see any o/p at AD9361. Can you please help?

mbr0wn, 2016-06-15:
This sounds more like a support request. Can you please post it to the mailing list?

Also, I'm not sure what you're trying to do. When you post to the list, please provide some details on what you want to achieve. Thanks!

closed 2016-06-15 by mbr0wn

mayuri2016, 2016-06-15:
Hi

Can u let me know the mailing list?

Regards
Mayuri

"DISCLAIMER: This message is proprietary to Aricent and is intended solely for the use of the individual to whom it is addressed. It may contain privileged or confidential information and should not be circulated or used for any purpose other than for what it is intended. If you have received this message in error, please notify the originator immediately. If you are not the intended recipient, you are notified that you are strictly prohibited from using, copying, altering, or disclosing the contents of this message. Aricent accepts no responsibility for loss or damage arising from the use of the information transmitted by this email including damage from virus."

mbr0wn, 2016-06-15:
https://www.ettus.com/support