EttusResearch/uhd #64: OAI eNB with USRP B210: DL PRB = 25 (BW = 5 MHz) is normal, DL PRB = 6 ( BW = 1.4 MHz) often display "L" & "U" on terminal

davidyu1, 2016-06-27:
Hi Ettus,
When we used OAI eNB with USRP B210: 
Execution on DL PRB = 25 (BW = 5 MHz) is normal, 
but is abnormal on DL PRB = 6 ( BW = 1.4 MHz) is anormal and It often display "L" & "U" on terminal.
Do you know why it cause this issue?

Regards,
David Yu

davidyu1, 2016-06-27:
Paste the log as follows:

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.009.004-release

# /dev/cpu_dma_latency set to 0us
log init done
Running with XFORMS!
num component carrier 1 

----------------------------------------------------------------------
 ENB CONFIG FILE CONTENT LOADED (TBC):
----------------------------------------------------------------------
ENB CONFIG for instance 0:

	eNB name:           	eNB_Eurecom_LTEBox:
	eNB ID:             	3584:
	Cell type:          	CELL_MACRO_ENB:
	TAC:                	1:
	MCC:                	208:
	MNC:                	93:
	eutra band for CC 0:         	5:
	downlink freq for CC 0:      	881500000:
	uplink freq offset for CC 0: 	-45000000:

	Cell ID for CC 0:	0:
	N_RB_DL for CC 0:	6:
	nb_antennas_tx for CC 0:	1:
	nb_antennas_rx for CC 0:	1:
	rach_numberOfRA_Preambles for CC 0:	15:
	rach_preamblesGroupAConfig for CC 0:	0:
	rach_powerRampingStep for CC 0:	2:
	rach_preambleInitialReceivedTargetPower for CC 0:	6:
	rach_preambleTransMax for CC 0:	6:
	rach_raResponseWindowSize for CC 0:	7:
	rach_macContentionResolutionTimer for CC 0:	5:
	rach_maxHARQ_Msg3Tx for CC 0:	5:
	bcch_modificationPeriodCoeff for CC 0:	0:
	pcch_defaultPagingCycle for CC 0:	2:
	pcch_nB for CC 0:	2:
	prach_root for CC 0:	0:
	prach_config_index for CC 0:	0:
	prach_high_speed for CC 0:	0:
	prach_zero_correlation for CC 0:	1:
	prach_freq_offset for CC 0:	2:
	pdsch_referenceSignalPower for CC 0:	-45:
	pdsch_p_b for CC 0:	0:
	pusch_n_SB for CC 0:	1:
	pusch_hoppingMode for CC 0:	0:
	pusch_hoppingOffset for CC 0:	0:
	pusch_enable64QAM for CC 0:	0:
	pusch_groupHoppingEnabled for CC 0:	1:
	pusch_groupAssignment for CC 0:	0:
	pusch_sequenceHoppingEnabled for CC 0:	0:
	pusch_nDMRS1 for CC 0:	1:
	pucch_delta_shift for CC 0:	0:
	pucch_nRB_CQI for CC 0:	1:
	pucch_nCS_AN for CC 0:	0:
	srs_enable for CC 0:	0:
	pusch_p0_Nominal for CC 0:	-60:
	pucch_p0_Nominal for CC 0:	-96:
	pusch_alpha for CC 0:	7:
	pucch_deltaF_Format1 for CC 0:	2:
	pucch_deltaF_Format1b for CC 0:	1:
	pucch_deltaF_Format2 for CC 0:	1:
	pucch_deltaF_Format2a for CC 0:	1:
	pucch_deltaF_Format2b for CC 0:	1:
	msg3_delta_Preamble for CC 0:	6:
	ul_CyclicPrefixLength for CC 0:	0:
	ue_TimersAndConstants_t300 for CC 0:	5:
	ue_TimersAndConstants_t301 for CC 0:	5:
[... 155 lines cut ...]
[S1AP][I][s1ap_eNB_handle_s1_setup_response] servedPLMNs.list.count 1
[ENB_APP][I][eNB_app_task] [eNB 0] Received S1AP_REGISTER_ENB_CNF: associated MME 1
Scope thread has priority 2
Scope thread created, ret=0
Setting eNB_thread FIFO scheduling policy with priority 99 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread0] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread1] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread2] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread3] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread4] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread5] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread6] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread7] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread8] 
Setting OS scheduler to SCHED_FIFO for eNB [cc0][thread9] 
Creating main eNB_thread 
[SCHED][eNB] Started eNB main thread on CPU 0 TID 4889
eNB_thread: mlockall in ...
eNB_thread: mlockall out ...
waiting for sync (eNB_thread)
Sending sync to all threads
TYPE <CTRL-C> TO TERMINATE
Entering ITTI signals handler
got sync (eNB_thread)
LULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULU [... cut ...] ULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULULLULLULLULLULLULLULLULLULLULLULLULLULULULULULULULULULULULUL

mbr0wn, 2016-06-27:
@davidyu1 This seems more like an OAI issue. I would recommend posting on their mailing list.

closed 2016-06-27 by mbr0wn