EttusResearch/uhd #65: OpenBTS - UHD incompatibility? (B200 mini)

RuxandraO, 2016-06-27:
Hi!

Is there any incompatibility between OpenBTS (run over Ubuntu 14.04) and UHD 3.9 or higher?

If I run OpenBTS, the devices can't properly connect to the network (even if on the phone it shows it's connected): the tmsis table remains always empty and nothing works (sms, voice, etc.).
I have this problem now when I try to run OpenBTS on USRP B200 mini from Ettus, but I had the same problem when I tried to run it on B210. At that point I could downgrade UHD to 3.7, but now I can't, because B200 mini only supports UHD 3.9 or higher.(note that I can see the device if I run uhd_find_devices or run ./transceiver)

Any solution to this? Has anybody successfully run OpenBTS on B200 mini? If yes, under what configuration?

Thanks!
Ruxandra

[P.S.] I've posted the same issue of the OpenBTS forum also, but as I did not had an answer yet, maybe you could help. Thanks!

mbr0wn, 2016-06-27:
@RuxandraO Sorry, we're not experts with OpenBTS. We do subscribe and post to the OpenBTS mailing list, so if you posted your issue there, that's a much better place to discuss it. I'll close this, since we don't really know if this is even a UHD issue.

closed 2016-06-27 by mbr0wn

Robert-steve, 2016-08-22:
Hi,
this problem has appeared since UHD 3.9 where the USRP B2XX does not work with OpenBTS because of a timing issue.

ttsou, 2016-08-30:
@Robert-steve wrote
>Hi,
this problem has appeared since UHD 3.9 where the USRP B2XX does not work with OpenBTS because of a timing issue.

The timing issue was addressed at the time of UHD 3.9.0 release.

Specifically note that the version check for UHD occurs during configuration *before* the build. The command './configure --with-uhd' must be run between builds. If the build is not reconfigured, the improper timing values will be compiled, though the transceiver will otherwise still operate normally.

https://github.com/RangeNetworks/openbts/commit/42cc2d123d9

Users of OpenBTS Ubuntu based bash scripts should verify that the above reconfiguration occurs between UHD version changes. Running UHD portions of the scripts in full is advisable.

Robert-steve, 2016-09-03:
> RangeNetworks/openbts@42cc2d1

Hi,

Should it also work for OpenBSC ?

ttsou, 2016-09-06:
@Robert-steve wrote:
>Should it also work for OpenBSC?

For use with the Osmocom GSM stack, the osmo-trx transceiver should be used. Note that osmo-trx is compatible with OpenBTS and Osmocom, but the OpenBTS transceiver is not compatible with Osmocom.

http://openbsc.osmocom.org/trac/wiki/OsmoTRX