EttusResearch/uhd #69: Ubuntu 14.04 and Centos 6 failure

sgardner340, 2016-08-14:
This started on Aug 04
 Current step: (uhd :: make)
make
bashexec (/home/hudson/workspace/PyBombs_Build_All_CentOS_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/src-rpm/src/uhd/host/build)::
find . -name '*.make' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/'
find . -name 'Makefile' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b\3\x1b\"/'
ln -s /home/hudson/workspace/PyBombs_Build_All_CentOS_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/cc
ln -s /home/hudson/workspace/PyBombs_Build_All_CentOS_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/gcc
ln -s /home/hudson/workspace/PyBombs_Build_All_CentOS_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/c++
ln -s /home/hudson/workspace/PyBombs_Build_All_CentOS_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/src-rpm/colorgcc.pl /apps/smokey64/bin/g++
make -j4
sed: no input files
sed: no input files
make: *** No targets specified and no makefile found. Stop.
bash return val = 2

mbr0wn, 2016-08-15:
@sgardner340 Can you please confirm you're running compilers, SWIG, Boost, etc. that are supported by your version of UHD? An unmodified CentOS 6 does not support UHD 3.10 and newer, IIRC. However, you should be getting good error messages if that's the case the first time you run CMake.

mbr0wn, 2016-08-16:
@sgardner340 Also, how are you building? I don't recognize any of those commands.

sgardner340, 2016-08-16:
I am building with the old Pybombs and my CentOS 6 system have not been
modified because I have to match my teams environment. Perhaps I need to
gitrev: an earlier version until we update our environment.

mbr0wn, 2016-08-16:
That might do the trick. I'm assuming there's nothing we can do here for now, ping this thread if you feel there's more that can be fixed.

closed 2016-08-16 by mbr0wn