EttusResearch/uhd #72: E300 SPI code incompatible with older kernels

rajb245, 2016-09-23:
The E300 SPI code does stuff like this:

```
struct spi_ioc_transfer tr; // found in <linux/spi/spidev.h>
...
tr.tx_nbits = 1;
tr.rx_nbits = 1;
```

This was updated from previous behavior at this time:
https://github.com/EttusResearch/uhd/commit/31d5c223ed6e0ca5569042fa1dfbcd0f18194d86

However, tx_nbits and rx_nbits were added to that struct around Linux 3.15:
https://github.com/torvalds/linux/commit/dc64d39b54c1e9db97a6fb1ca52598c981728157

On the other hand, Ubuntu 14.04 is an LTS release that shipped with Linux 3.13 and has been getting bugfixes and patches on this kernel since it's release, so E300 code is broken at least on that platform if users don't upgrade their kernel manually. Canonical don't intend to stop supporting this [until 2019](https://wiki.ubuntu.com/Releases).

If kernels older than 3.15 are to be supported, then [patch attached](https://github.com/EttusResearch/uhd/files/490590/e300_spi_dual_quad_fix_for_old_kernels.diff.txt) (don't feel like cloning all of UHD on github to submit these few lousy lines as a pull req.). This should probably go into maint and master.

If kernels older than 3.15 are not supported, this fact should be documented [here](https://files.ettus.com/manual/page_build_guide.html) under Linux notes as a minimum kernel version.

mbr0wn, 2016-09-23:
@mfischer

mfischer, 2016-09-26:
We generally don't support running distros / kernels different from the one we shipped. While I'm fine with your patch (not a fan of the #ifdefs, though) I doubt it's the only thing required to make UHD run on Ubuntu.
I think we should document that we don't officially support Kernels other than the now (ancient) 3.14 Xilinx tree.
@rajb245 to not be all doom and gloom we got improvements in the pipeline for next-gen products that we'll backport that will get us much closer to a (modern) mainline.

rajb245, 2016-09-26:
I guess I should mention that this is not purely an embedded issue; UHD fails to compile on anything with an old kernel (e.g., my desktop) when configured with -DENABLE_E300=ON. Whether a user should enable this flag or not on their desktop builds is another question, but if they do, they'll be puzzled by the compiler error complaining that the struct doesn't have fields named "tx_nbits" and "rx_nbits". The patch prevents that scenario and doesn't hurt things on actual embedded builds.

mbr0wn, 2016-09-26:
@rajb245 Thanks for the info. We can see how this can cause issues in some circumstances, but we do provide `-DE300_FORCE_NETWORK=ON` specifically for this purpose, and we don't want to start having kernel-version specific hacks in our code as much as possible.

Native builds (or cross-compiles) don't care either way, and network mode should only be enabled if the user really knows what they're doing -- in which case `-DE300_FORCE_NETWORK=ON` is the right option.

Long story short, `E300_NATIVE` should not be defined if you're compiling for network mode. If it was, that would be an actual bug we need to fix.

closed 2016-09-26 by mbr0wn

rajb245, 2016-09-26:
@mbr0wn thanks for clarifying this. If -DENABLE_E300=ON is not required on a "host" that interacts with an E310 in network mode, then the documentation [here]http://files.ettus.com/manual/page_usrp_e3x0.html#e3x0_uhd_build) should be updated (or clarified in the case that I was reading it incorrectly to begin with).

mbr0wn, 2016-09-26:
@rajb245 Not a problem. Here's our current manual:
> To work with your E-Series device in network mode, you will need to build UHD on your PC with extra
> CMake flags. Assuming you are in the host/build directory, see below:
>
>    $ cmake -DENABLE_E300=ON -DE300_FORCE_NETWORK=ON ..
>    $ make

If you have suggestions how to clarify these lines, they're appreciated! I do believe it adds all the necessary info, though.

rajb245, 2016-09-27:
@mbr0wn 
That says that -DENABLE_E300=ON should be set on a host (e.g., my desktop PC) interacting with an E300 series device in network mode, correct?

But UHD in that configuration will not compile on Linux < 3.15, as I originally reported. So...this is still a bug I think. Scenario:
I get an E310 and want to use it in networ mode so download an Ubuntu LTS release, download UHD, follow the UHD documentation for compiling, hit a cryptic compiler error about the spi_ioc_transfer structure, and conclude that the UHD driver is broken and Ettus is shipping a broken product.

This may affect a very small portion of people working with UHD (chances are I'm the only person to come across this), so maybe it's moot and the bug should stay closed. In that case, I would recommend a note in the documentation saying that kernels older than 3.15 don't support any build with the -DENABLE_E300=ON flag enabled.

mbr0wn, 2016-09-27:
@rajb245 Take another look: It says you need *both* ` -DENABLE_E300=ON` and  `-DE300_FORCE_NETWORK=ON`.

rajb245, 2016-09-28:
@mbr0wn Okay, I can verify that compiling with `-DENABLE_E300=ON` and `-DE300_FORCE_NETWORK=ON` does not cause `E300_NATIVE` to be defined; I had mistakenly thought that that any build with `ENABLE_E300=ON` would set `E300_NATIVE`, but I see that's not the case. Consider this matter closed and thanks for your help.