EttusResearch/uhd #75: Unable to compile tx_waveforms

kirtyvedula, 2016-10-06:
Trying to use make to compile tx_waveforms.cpp

The following error creeps up.

/home/usrp/tx_waveforms/tx_waveforms.cpp:18:25: fatal error: wavetable.hpp: No such file or directory
 #include "wavetable.hpp"
                         ^
compilation terminated.
CMakeFiles/tx_waveforms.dir/build.make:54: recipe for target 'CMakeFiles/tx_waveforms.dir/tx_waveforms.cpp.o' failed
make[2]: *** [CMakeFiles/tx_waveforms.dir/tx_waveforms.cpp.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/tx_waveforms.dir/all' failed
make[1]: *** [CMakeFiles/tx_waveforms.dir/all] Error 2
Makefile:72: recipe for target 'all' failed
make: *** [all] Error 2

primercuervo, 2016-10-06:
It seems that you are trying to compile tx_waveforms by locating it outside the uhd/host/examples/ directory, which is completely fine if that suits your needs. 

Look that it is telling you that the compiler can't find the wavetable. You can either have the tx_waveforms located at uhd/host/examples/ and compile the whole uhd in a build directory that you create at /uhd/host/, or copy the wavetable to the directory in which you intend to compile.

The missing file is the following:
https://github.com/EttusResearch/uhd/blob/master/host/examples/wavetable.hpp

primercuervo, 2016-10-06:
There doesn't seem to be an issue with the code base. Please let us know if you overcome this problem. I suggest you take usage issues to the mailing list, where further discussion can take place with several expert readers.

mbr0wn, 2016-10-06:
How did you trigger this build?

mbr0wn, 2016-10-31:
Since `tx_waveforms` compiles just fine on regular UHD, I'm closing this.

closed 2016-10-31 by mbr0wn