EttusResearch/uhd #79: request for yakkety on the ppa

guruofquality, 2016-11-30:
Hi, Im using the [uhd ppa](https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd) as a dependency for [myriadrf drivers ppa](http://wiki.myriadrf.org/Packaging#Drivers_PPA). But the uhd is missing the package for the latest ubuntu (yakkety), and the official ubuntu packages are a version behind.

mbr0wn, 2016-11-30:
Absolutely. Thanks for pinging us on this. I've added yakkety to the list of versions that get automatically included, so the next release will already get it. I'll run a build on the latest version in a bit.

csete, 2016-11-30:
Btw. a heads up wrt UHD 3.9 and Boost on Ubuntu 16.10:
https://bugs.launchpad.net/ubuntu/+source/gqrx-sdr/+bug/1637774

Hopefully it will no longer be an issue with UHD 3.10

mbr0wn, 2016-11-30:
Well, ain't that fantastic. Thanks for the heads-up.

guruofquality, 2016-12-17:
@mbr0wn any update on this?

dkozel, 2016-12-19:
@guruofquality The PPA is all set and will include 16.10 on the next build. Unfortunately Launchpad provides no way to rebuild or extend an existing release, so until we have the next release ready there's no way of adding the support.

guruofquality, 2016-12-19:
I'm not sure what the technical issue was. But while you cant fix an existing failed upload you can reupload a fixed source package with an incremented version number. Depending upon how things were changed for the fix:

1) If the source tarball didnt change, you can usually bump the second part of the version number in some way (after the dash) to get a new build. so `uhd - 3.10.1.0-0ubuntu1~xenial1 ` could become `uhd - 3.10.1.0-0ubuntu1~xenial2 `

2) If for some reason the source tarball changed (sometimes different tools tar up different bits for the same tag), you can just make up a version number after the patch number to append to the end. so `uhd - 3.10.1.0-0ubuntu1~xenial1 ` could become `uhd - 3.10.1.0.1-0ubuntu1~xenial1 `

At least thats my experience from maintaining myriadrf ppa. I'm happy to help debug. Not an expert or anything, just have my share of experiences goofing up with the launchpad ppas.

I might try to get this uhd release up on myriadrf drivers(at least for yakkety) since its a dependency for some other things that I want to upload. Any advice ahead of time on what went wrong?

mbr0wn, 2017-01-03:
So, I did actually re-upload the code and had assumed launchpad would build the yakkety package (since it didn't already have a version of UHD). Not sure what went wrong. We should be able to add a yakkety build without bumping versions. Does Launchpad require activating certain distros?

guruofquality, 2017-01-04:
@mbr0wn I made an attempt to upload uhd myself on the myriadrf/drivers for yakkety twice now and I had a similar problem. It never built and I didnt get a failure notification. And you shouldn't have to activate any specific release. I have uploaded other drivers for yakkety just fine. :-/

No idea whats going on. Did you get any launchpad accept/reject emails? I stopped getting them recently, though other builds have of course worked, but I still get emails for build failures.

So that's worry-some, let me know if you figure anything out. I will do the same.

mbr0wn, 2017-01-04:
That's exactly what I got. No response, no packages.

dkozel, 2017-01-30:
Yakkety packages are now on the PPA for UHD 3.10.1.1
https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd

closed 2017-01-30 by dkozel

csete, 2017-01-31:
Thanks for the packages. Unfortunately, this version also crashes with the Boost that comes with Ubuntu 16.10.

dkozel, 2017-02-02:
@csete We've tried a clean install of the packages from the PPA on a fresh Ubuntu 16.10 machine and were able to run a variety of included examples and utilities with it with no problem. Can you provide more information about the crash you're observing?

csete, 2017-02-03:
@dkozel Sorry for the noise... @guruofquality had the same experience as you, so it seems that it is only an issue with gr-osmosdr. We are looking into it.

dkozel, 2017-02-03:
@csete No worries, Please always feel free to let us know about errors you encounter which might be caused by UHD. GQRX is great software, thank you for everything you've done.

guruofquality, 2017-02-03:
@dkozel We were having some discussion here: https://discourse.myriadrf.org/t/packages-for-ubuntu-16-10/667

I can replicate the issue. UHD itself seems fine, but it seems to be segfauling with gr-osmosdr. I built everything from source (except uhd from the ppa) and I disabled everything in gr-osmosdr except the uhd support and ran osmocom_fft and it crashes with some asio traceback from with uhd just like in the launchpad bug report. I suspect its something to do with both uhd and gnuradio blocks using boost asio. But I didnt test any deeper.

Do gr-uhd apps for work you? What about osmocom_fft?