EttusResearch/uhd #80: Homebrew-installed uhd fails to download images.

tee3, 2016-12-01:
I am using the Homebrew-installed `uhd` formula and so far things work fairly well.  Homebrew is well-liked in the macOS community and not having to use MacPorts is a big positive for me.

I fixed this problem by running `pip install requests` as the user that I use for installing packages via Homebrew, which fixed the issue and the images were downloaded successfully.  Note also that the firmware update program worked successfully once the images were downloaded.

The main issue is that `$(brew --prefix uhd)/lib/uhd/utils/uhd_images_downloader.py` requires that the Python `requests` package is installed.  

How is this handled on Linux?  Is it installed as a dependency via the package manager?  Homebrew does not encourage installing Python packages, but maybe add a caveat that suggests a `pip` command line to install all required packages?

What do you think?

mbr0wn, 2016-12-01:
@michaelld Any suggestions? I know that we check for requests on source builds.

@tee3 On Linux, if you source build, UHD won't build unless you have requests. If you install through the package manager, requests is a dependency. Windows is more like your case; and there, we just print an error message if it fails.

michaelld, 2016-12-01:
@tee3 : what brew recipe (or whatever that's called) did you use? UHD does indeed require Python requests, as you say. In MacPorts-land, py*-requests is a build dependency, so it's installed before UHD is built. I'd guess one could update the Brew script for UHD to include this dependency; that's more of a Seth thing though.

@mbr0wn maybe tag Seth here (I can't seem to do it)?

tee3, 2016-12-01:
@mbr0wn Yes, I understand that and I've done it both ways.  However, I really like installing it using the package manager :)

@michaelld It is actually a part of `homebrew-core`, so `brew install uhd` is what I used.  I figured you guys had created the formula, but now I'm not sure so.  The formula is here: https://github.com/Homebrew/homebrew-core/blob/master/Formula/uhd.rb

If the the actual build process (`cmake`, `make`) actually does is supposed to install those dependencies, it should "just work" and install what's required on the system.  That's why I was confused since the formula just in essence runs `cmake .. && cd .. && make install` on the tagged version.

After looking at the formula, it looks like it puts the Python requirements into a vendored location.  Maybe it just doesn't look there when it is run?

mbr0wn, 2016-12-01:
@tee3 `cmake && make` won't install requests, but it'll check if it's there.

mbr0wn, 2016-12-01:
@sdh11 Might have some insight (he doesn't work for Ettus, but has them Mac OS smarts). Quite honestly, I have no clue what the right answer is here, and I'm not sure anyone on the Ettus has if @michaelld doesn't.

mbr0wn, 2016-12-01:
Oh, and I have no clue who wrote that Formula. Until you posted this, I had actually never looked at such a file before :astonished: 

It doesn't seem to include requests at all. Maybe you can ping the author and ask her to include it?

tee3, 2016-12-01:
@mbr0wn Thanks, I will take it up with the Homebrew team.  Thanks for all your insights!

sdh11, 2016-12-02:
@tee3 It looks like the method for installing `pip` packages from homebrew is with resources;`mako` seems to be installed this way.

https://github.com/Homebrew/homebrew-core/blob/master/Formula/theharvester.rb has an example of installing `requests` from the recipe, but I'm not sure how this works at the moment.

sdh11, 2016-12-02:
Also, a limitation of installing `requests` this way is that it is not available system wide.

mbr0wn, 2017-01-04:
This seems like a non-UHD-specific issue. We should figure out how to do this, but I'm closing this specific issue for now.

closed 2017-01-04 by mbr0wn