EttusResearch/uhd #81: nocscript_parser_test fails on ppc64

yarda, 2016-12-02:
The nocscript_parser_test fails with uhd-3.10.1.0 (and also with the current git head) on ppc64 (rhel-7):

32/32 Test #32: nocscript_parser_test ............***Failed    0.02 sec

97% tests passed, 1 tests failed out of 32

By running it manualy::
./nocscript_parser_test_test.sh 
linux; GNU C++ version 4.8.5 20150623 (Red Hat 4.8.5-11); Boost_105300; UHD_003.010.000.HEAD-304-g3bf4b000

Running 6 test cases...
unknown location(0): fatal error in "test_adds_no_vars": std::runtime_error: SyntaxError: Parsing stopped at: ADD(1, ADD(2, ADD(3, 4)))
Error message: 
/root/uhd/host/tests/nocscript_parser_test.cpp(90): last checkpoint
unknown location(0): fatal error in "test_adds_with_vars": std::runtime_error: SyntaxError: Parsing stopped at: ADD(1, ADD(2, $spp))
Error message: 
/root/uhd/host/tests/nocscript_parser_test.cpp(90): last checkpoint
unknown location(0): fatal error in "test_fft_check": std::runtime_error: SyntaxError: Parsing stopped at: GE($spp, 16) AND LE($spp, 4096) AND IS_PWR_OF_2($spp)
Error message: 
/root/uhd/host/tests/nocscript_parser_test.cpp(90): last checkpoint
unknown location(0): fatal error in "test_pure_string": std::runtime_error: SyntaxError: Parsing stopped at: 'foo foo', "bar"
Error message: 
/root/uhd/host/tests/nocscript_parser_test.cpp(90): last checkpoint
unknown location(0): fatal error in "test_multi_commmand": std::runtime_error: SyntaxError: Parsing stopped at: DUMMY(), DUMMY(), DUMMY()
Error message: 
/root/uhd/host/tests/nocscript_parser_test.cpp(90): last checkpoint

*** 5 failures detected in test suite "Master Test Suite"

I bisected the problem to the following commit:
3bf4b000f7d9a7f4af82c21753556ede7e8df6e3

But unfortunately the commit is huge, I haven't time to look more deep on it. It doesn't fail on x86.

yarda, 2016-12-02:
It seems it isn't regression, the failing test cases and functionality just wasn't there. The commit 3bf4b00 is adding it. Maybe the problem is in create_expr_tree? Hopefully I will look on it deeper during next week.

mbr0wn, 2016-12-02:
@yarda Thanks a lot for testing and bisecting. We don't currently have a ppc64 build in our CI system, so we'll have to try and replicate it.

yarda, 2016-12-06:
It could be boost error, lex::tokenize is failing for e.g. "ADD(1, 2)".

mbr0wn, 2016-12-06:
A Boost error seems most likely.

yarda, 2016-12-07:
Yes, it's definitely boost, Red Hat bugzilla:
https://bugzilla.redhat.com/show_bug.cgi?id=1402516

Boost tracker including the patch:
https://svn.boost.org/trac/boost/ticket/8291

So feel free to close this bug as invalid, but some more questions follows. We would like to rebase UHD in RHEL/EPEL but fixing this bug in RHEL boost could take some time. I wonder whether the nocscript parser is used only by RFNoC? Is there a easy way how to disable it / drop this functionality (instead of reverting the commit)?

yarda, 2016-12-08:
Unfortunately it doesn't seem trivial to drop the parser (and RFNoC) from the uhd-3.10.1.0 and I don't want to support such modified build in the RHEL thus deferring the uhd rebase until boost get fixed.

Sorry for invalid report.

closed 2016-12-08 by yarda

mbr0wn, 2016-12-08:
@yarda this was a great report, and thanks for all the detailed work you did to discover this. Yeah, I'm afraid dropping RFNoC from 3.10 would mean disabling the X310 and X300 devices. This is not the first time we're having troubles with Boost versions, but in this case, there's not a lot we can do.