EttusResearch/uhd #87: Compilation error on lubuntu 16.10

brchiu, 2017-01-06:
Since currently uhd driver is not available for distribution yakkety on the ppa, I tried to build it on my own.
However I got the following error message during compilation. 

```
[ 61%] Built target uhd
[ 61%] Linking CXX executable sync_to_gps
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail_106100::cpp_regex_traits_implementation<char>::transform_primary(char const*, char const*) const'
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail_106100::cpp_regex_traits_implementation<char>::transform(char const*, char const*) const'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/sync_to_gps.dir/build.make:107: recipe for target 'examples/sync_to_gps' failed
make[2]: *** [examples/sync_to_gps] Error 1
CMakeFiles/Makefile2:414: recipe for target 'examples/CMakeFiles/sync_to_gps.dir/all' failed
make[1]: *** [examples/CMakeFiles/sync_to_gps.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

I have checked that the version of  `libboost_regex.so` is correct by using `nm -D --demangle libboost_regex.so` to list these class methods. Moreover, I tried to use content of `link.txt` created under directory `sync_to_gps.dir` to perform link manually but it does not work.

I have no idea about why linking failed.

brchiu, 2017-01-06:
By the way, since the newest version libboost on yakkety is 1.61, I slightly modify CMakeLists.txt locally to accept boost 1.61.

mbr0wn, 2017-01-06:
I just tried building UHD on a 16.10 Docker, and it built cleanly. It picked up Boost 1.61 without modifications to CMake, although I'll put that in just in case.

Which branch did you build? And was this a fresh install of everything?

mbr0wn, 2017-01-06:
FYI I tried both maint and master on my Docker.

mbr0wn, 2017-01-07:
I tried a lubuntu image specifically, and that also worked. @brchiu please take this to the mailing list -- I'll close this for now, but will happily reopen once we know this is an actual UHD bug.

closed 2017-01-07 by mbr0wn

brchiu, 2017-01-07:
@mbr0wn, 

Thanks for your prompt response.

I perhaps found the preliminary cause of my encountered problem.

The boost libraries were compiled with new ABI, so when I use `nm -D --demangle` to list its class method inside libraries, there are **[abi:cxx11]** tag in some class method name. For example : 

```
boost::re_detail_106100::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const 
```
However, the linker is looking for a class method name without [abi:cxx11], so it failed.

```
boost::re_detail_106100::cpp_regex_traits_implementation<char>::transform_primary(char const*, char const*) const
```

I surfed internet and learned that this new ABI is introduced after **gcc-5**. But I already use **gcc-5.4.1** to compile UHD in this case. I don't know how instruct GCC to look for [abi:cxx11]-mangled class method name.

Interestingly, since I have **gcc-6** installed on my laptop, so I tried to use gcc-6 for compilation.
Then **It compiled** without changing any thing. 

ps. My laptop was upgraded from Lubuntu 16.04, it might be the reason that why it stayed at gcc-5 as default version.

mbr0wn, 2017-01-10:
If you use Clang, you would have the same issue. Good this is cleared up!

annaRadiant, 2017-02-20:
I am getting the same issue ... how exactly did you solve this?

Linking CXX executable sync_to_gps
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail_106100::cpp_regex_traits_implementation<char>::transform_primary(char const*, char const*) const'
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail_106100::cpp_regex_traits_implementation<char>::transform(char const*, char const*) const'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/sync_to_gps.dir/build.make:106: recipe for target 'examples/sync_to_gps' failed
make[2]: *** [examples/sync_to_gps] Error 1
CMakeFiles/Makefile2:414: recipe for target 'examples/CMakeFiles/sync_to_gps.dir/all' failed
make[1]: *** [examples/CMakeFiles/sync_to_gps.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
UHD build apparently failed
Exiting UHD build

brchiu, 2017-02-20:
@annaRadiant   I use **gcc-6** to compile and success without linking error.

annaRadiant, 2017-02-20:
I was already running 

 sudo apt-get install gcc-6 g++-6
Reading package lists... Done
Building dependency tree       
Reading state information... Done
g++-6 is already the newest version (6.2.0-7ubuntu11).
g++-6 set to manually installed.
gcc-6 is already the newest version (6.2.0-7ubuntu11).
gcc-6 set to manually installed.
The following packages were automatically installed and are no longer required:
  bridge-utils containerd runc ubuntu-fan
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 145 not upgraded.

brchiu, 2017-02-20:
@annaRadiant Check your **default** gcc's version by executing **gcc --version**. If it is not **gcc-6**, for example **gcc-5** due to legacy installation, you might need to use **update-alternatives** to select **gcc-6** as your default version.

noc0lour, 2017-02-20:
Ubuntu 16.10 should have gcc 6 enabled by default since the system libraries are built with gcc-6. I just verified and on "new" Ubuntu 16.10 images gcc 6 is used by default.

annaRadiant, 2017-02-20:
@noc0lour ... @brchiu was correct. The default I was still running was gcc 5.4, so I had to update alternatives. I installed 16.10 last week. I am rerunning the build gnuradio script now to see if the problem corrected itself. I will let you know.

annaRadiant, 2017-02-20:
Yeah ... it worked

brchiu, 2017-02-21:
@noc0lour Ubnuntu 16.10 is shipped with gcc-6 enabled as default, however, some people, like me and @annaRadiant maybe the case, we upgrade our system from previous Ubuntu version with gcc-5 as default. So we encounter this problem. People use a cleanly installed Ubuntu 16.10 will not be aware of it.

haidfs, 2018-10-26:
@brchiu Thanks for your tips, but it didn't work in my environment.abi-cxx11,it trapped me for so long.I added this option**-DCMAKE_CXX_FLAGS=-D__GLIBCXX_USE_CXX11_ABI=0** after my cmake order, removed the cache, and it worked! You can get more from the following url：https://github.com/caffe2/caffe2/issues/1980