EttusResearch/uhd #88: dissectors wireshark API changes

selinux, 2017-02-01:
when I start wireshark 2.2.3 after succesfully build wireshark dissectors I get an...
chdr-plugin.so: undefined symbol tvb_get_string

It seems that tvb_get_string() has been replaced by tvb_get_string_enc() 

from wireshark...

=== Major API Changes
* tvb_get_string and tvb_get_stringz have been replaced with
tvb_get_string_enc and tvb_get_stringz_enc respectively.

mbr0wn, 2017-02-02:
@selinux Thanks for the pointer! We'll need to make sure we support a spectrum of Wireshark APIs, but this seems fixable.

mbr0wn, 2017-06-21:
I remember seeing this, too, but I can't reproduce now. I have the dissector running on 2.2.6. Not sure if some changes were required, but I'm closing unless we find a way to reproduce this.

closed 2017-06-21 by mbr0wn