EttusResearch/uhd #89: make fpga git hashes visible on the host

aholtzma, 2017-02-10:
In 3.10.x the x300/e300/n230 FPGAs all have the git hash baked into them, but there is no way to get this information out to the host. Knowing exactly what FPGA is loaded is useful in tracking down bugs, as well as knowing if the FPGA needs to be updated. Even though UHD has the concept of 'compatibility', in practice this means you have to do system testing with all possible 'compatible' FPGA builds which is untenable. 

I propose adding this hash value as a node in the property tree.

dkozel, 2017-02-10:
Hello @aholtzma 

Happy to say that readback of the FPGA hash is already available!
`uhd_usrp_probe --string /mboards/0/fpga_version_hash`

https://github.com/EttusResearch/uhd/commit/7998db839f5b1515b24dcba5ecae6607685dce99

aholtzma, 2017-02-10:
Great! How about printing it in the headline if it exists, so you always have that data for debugging?

Something like:

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.000.HEAD-303-gc5b07617;  FPGA_XXXXXXXXXX

mbr0wn, 2017-02-10:
The headline is generated independent of the device, but we can (and should) add this to `uhd_usrp_probe`.

aholtzma, 2017-02-10:
Ok fair enough. One last suggestion is to print a warning if the FPGA hash doesn't match the released FPGA hash. This will flag cases where people are having trouble using the latest UHD with a FPGA from a point release earlier (which is going to be a lightly-if-at-all tested scenario). While the FPGAs are in theory compatible, small changes in behaviour may break things. Knowing the FPGA is different upfront saves a lot of debugging time.

mbr0wn, 2017-02-10:
We actually have a fair amount of users who wouldn't be too happy with such a warning, because we explicitly allow the modification of the FPGA image.
There's a compromise here though: We're currently remodelling the logging infrastructure of UHD, and we could simply print the current hash with a lower log level, such that it doesn't always get printed unless you request a higher logging level for debugging purposes. That would make your debugging easier, without disturbing the people who want their modifications in.

I'll close this issue for now -- the git hash in `uhd_usrp_probe` should be released soon, and I'll track the printing of the git hash with a debug log level internally. Thanks for all the comments @aholtzma!

closed 2017-02-10 by mbr0wn