EttusResearch/uhd #90: Possible bug in function uhd_usrp_get_rx_info()

koksaki, 2017-02-15:
Hi,

I think I found bug in function **_uhd_usrp_get_rx_info_** located in file **_usrp_c.cpp_**.

Corresponding to the header file **_usrp_info.h_**, stucture **_uhd_usrp_rx_info_t_** is defined like this:

```c
typedef struct {
    //! Motherboard ID
    char* mboard_id;
    //! Motherboard name
    char* mboard_name;
    //! Motherboard serial
    char* mboard_serial;
    //! RX daughterboard ID
    char* rx_id;
    //! RX subdev name
    char* rx_subdev_name;
    //! RX subdev spec
    char* rx_subdev_spec;
    //! RX daughterboard serial
    char* rx_serial;
    //! RX daughterboard antenna
    char* rx_antenna;
} uhd_usrp_rx_info_t;
```

and the definition of function **_uhd_usrp_get_rx_info_** is:

```c
#define COPY_INFO_FIELD(out, dict, field) \
    out->field = strdup(dict.get(BOOST_STRINGIZE(field)).c_str())

uhd_error uhd_usrp_get_rx_info(
    uhd_usrp_handle h,
    size_t chan,
    uhd_usrp_rx_info_t *info_out
) {
    UHD_SAFE_C_SAVE_ERROR(h,
        uhd::dict<std::string, std::string> rx_info = USRP(h)->get_usrp_rx_info(chan);

        COPY_INFO_FIELD(info_out, rx_info, mboard_id);
        COPY_INFO_FIELD(info_out, rx_info, mboard_serial);
        COPY_INFO_FIELD(info_out, rx_info, rx_id);
        COPY_INFO_FIELD(info_out, rx_info, rx_subdev_name);
        COPY_INFO_FIELD(info_out, rx_info, rx_subdev_spec);
        COPY_INFO_FIELD(info_out, rx_info, rx_serial);
        COPY_INFO_FIELD(info_out, rx_info, rx_antenna);
    )
}
```

You can notice that field **_mboard_name_** will remain uninitialized. 

To solve this just add one line:

```c
#define COPY_INFO_FIELD(out, dict, field) \
    out->field = strdup(dict.get(BOOST_STRINGIZE(field)).c_str())

uhd_error uhd_usrp_get_rx_info(
    uhd_usrp_handle h,
    size_t chan,
    uhd_usrp_rx_info_t *info_out
) {
    UHD_SAFE_C_SAVE_ERROR(h,
        uhd::dict<std::string, std::string> rx_info = USRP(h)->get_usrp_rx_info(chan);

        COPY_INFO_FIELD(info_out, rx_info, mboard_id);
        COPY_INFO_FIELD(info_out, rx_info, mboard_name);     /* Add this line to existing code */
        COPY_INFO_FIELD(info_out, rx_info, mboard_serial);
        COPY_INFO_FIELD(info_out, rx_info, rx_id);
        COPY_INFO_FIELD(info_out, rx_info, rx_subdev_name);
        COPY_INFO_FIELD(info_out, rx_info, rx_subdev_spec);
        COPY_INFO_FIELD(info_out, rx_info, rx_serial);
        COPY_INFO_FIELD(info_out, rx_info, rx_antenna);
    )
}
```

Same applies to function **_uhd_usrp_get_tx_info_**.

Ivan

mbr0wn, 2017-02-16:
Thanks @koksaki, you're right, and it'll be fixed on the next release (and before that on maint).

mbr0wn, 2017-02-17:
We have a fix in the pipeline.

closed 2017-02-17 by mbr0wn