EttusResearch/uhd #91: Possible inconsistency between files usrp_c.cpp and usrp.h

koksaki, 2017-02-18:
Hi,

I have noticed inconsistency when comparing **_uhd.lib_** import library and **_uhd.h_** header file.
I found that **_uhd.h_** header file declares functions **_uhd_usrp_get_rx_lo_names()_** and
**_uhd_usrp_get_rx_lo_sources()_**, but listing **_uhd.lib_** library I could not find references for
those two functions. This was verified also by linker, when I tried to build program that 
uses those two functions link process failed saying it could not find references for those functions.

Maybe I found explanation for this phenomenon:

Definitions of functions **_uhd_usrp_get_rx_lo_names()_** and **_uhd_usrp_get_rx_lo_sources()_**
are located in file **_usrp_c.cpp_**, and they are shown below:

```c
UHD_API uhd_error uhd_usrp_get_rx_lo_names(
    uhd_usrp_handle h,
    size_t chan,
    uhd_string_vector_handle rx_lo_names_out
){
    UHD_SAFE_C_SAVE_ERROR(h,
        rx_lo_names_out->string_vector_cpp = USRP(h)->get_rx_lo_names(chan);
    )
}

UHD_API uhd_error uhd_usrp_get_rx_lo_sources(
    uhd_usrp_handle h,
    const char* name,
    size_t chan,
    uhd_string_vector_handle rx_lo_sources_out
){
    UHD_SAFE_C_SAVE_ERROR(h,
        rx_lo_sources_out->string_vector_cpp = USRP(h)->get_rx_lo_sources(name, chan);
    )
}
```

On the other side, declarations of functions **_uhd_usrp_get_rx_lo_names()_** and
**_uhd_usrp_get_rx_lo_sources()_** are located in file **_usrp.h_**, and they are shown below:

```c
UHD_API uhd_error uhd_usrp_get_rx_lo_names(
    uhd_usrp_handle h,
    size_t chan,
    uhd_string_vector_handle *rx_lo_names_out
);

UHD_API uhd_error uhd_usrp_get_rx_lo_sources(
    uhd_usrp_handle h,
    const char* name,
    size_t chan,
    uhd_string_vector_handle *rx_lo_sources_out
);
```
By comparing definitions, given in file **_usrp_c.cpp_**, and declarations, given in file **_usrp.h_**,
you can notice type mismatch. Declaration is saying that argument
**_rx_lo_names_out/rx_lo_sources_out_** should be **_passed by reference_**,
but definition is saying that argument **_rx_lo_names_out/rx_lo_sources_out_**
should be **_passed by value_**. 

Obviously, argument **_rx_lo_names_out/rx_lo_sources_out_** should be **_passed by reference_**.

Question is -> **_How can something like this pass compilation process?_**

Ivan

mbr0wn, 2017-02-20:
@koksaki Thanks for pointing this out.

mbr0wn, 2017-02-20:
A fix was merged and will be available very soon.

closed 2017-02-20 by mbr0wn