EttusResearch/uhd #92: No UHD device after running build gnuradio script

annaRadiant, 2017-02-21:
I had difficulty with the build gnuradio script. I had to update GCC to 6, so I am running 6.2.0 on Ubuntu 16.10. 

It finished this morning, and now I am having problems finding the UHD devices. I have the GigE cable from my USRP N210 to the TP-Link router, with a cable to my dell latitude E6540 laptop. The router is getting a Fios connection. 

Here are the errors

$ ifconfig -a
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 0.0.0.0
        ether 02:42:f6:12:fd:2f  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eno1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.2  netmask 255.255.255.0  broadcast 192.168.10.255
        inet6 fe80::3643:64e2:434:ef84  prefixlen 64  scopeid 0x20<link>
        ether f0:1f:af:3e:be:61  txqueuelen 1000  (Ethernet)
        RX packets 3649  bytes 2782820 (2.7 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 25479  bytes 4441721 (4.4 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 20  memory 0xf7d00000-f7d20000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 611633  bytes 37156163 (37.1 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 611633  bytes 37156163 (37.1 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.166  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::f6c6:b878:c9cf:494  prefixlen 64  scopeid 0x20<link>
        ether b4:b6:76:cd:d4:8a  txqueuelen 1000  (Ethernet)
        RX packets 1189764  bytes 1307788880 (1.3 GB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 463045  bytes 50267803 (50.2 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  
collisions 0

$ uhd_usrp_probe --args="addr=192.168.10.2"
linux; GNU C++ version 6.2.0 20161018; Boost_106100; UHD_3.11.0.git-88-g180d5f3a

Error: LookupError: KeyError: No devices found for ----->
Device Address:
    addr: 192.168.10.2

$ uhd_find_devices
linux; GNU C++ version 6.2.0 20161018; Boost_106100; UHD_3.11.0.git-88-g180d5f3a

No UHD Devices Found

I ran populate_images.py to get the current images. 

Is there a reason it isn't seeing the UHD ... is the build gnuradio-script still having issues when trying to build the UHD drivers?

michael-west, 2017-02-21:
Both your host interface and the N210 have the same IP (192.168.10.2).  Set the IP address on your host interface to 192.168.10.1.

mbr0wn, 2017-02-21:
Also, you shouldn't have to run populate_images.py. `uhd_images_downloader` usually does all you want.

I'm closing this, it looks more like a case for the mailing list.

closed 2017-02-21 by mbr0wn