EttusResearch/uhd #94: race condition when launching 2 flowgraphs with b200s at the same time

osh, 2017-03-02:
File ".........../lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 2078, in make
    return _uhd_swig.usrp_sink_make(*args)
RuntimeError: AssertionError: libusb_claim_interface(this->get(), interface) == 0
  in virtual void libusb_device_handle_impl::claim_interface(int)
  at /usr/local/workspace/2016/src/uhd/host/lib/transport/libusb1_base.cpp:271

mbr0wn, 2017-03-03:
This is a legitimate issue, and we're looking into it.

osh, 2017-03-03:
It's actually quite an important use case, usually when I want my flowgraph samples synchronized to with 1ns I launch the two gr apps within 0.5ns of each other.

deanhuff, 2019-04-02:
it appears this race still exists as of 3.14 any updates?

manderseck, 2021-09-13:
@osh @deanhuff This issue seems to have been around for way too long. Therefore I checked if it still exists with the latest code from Git. Tests performed:
1. One instance of GNU Radio Companion with a flowgraph consisting of two USRP sources and two QT GUI Waterfall sinks in parallel for two USRPs B210. Works.
2. One instance of GRC with two different flowgraphs both consisting of a USRP source and a QT GUI Waterfall sink for two USRPs B210. Works.
3. Two instances of GRC with a flowgraph each consisting of a USRP source and a QT GUI Waterfall sink, again for two USRPs B210. Works, too.

Therefore I consider this issue as gone with the latest version. I'm really sorry that this issue was not handled earlier. Please re-open if you still encounter this issue with the latest UHD versions.

closed 2021-09-13 by manderseck