EttusResearch/uhd #98: uhd 3.9.6 tests fail

ZeroChaos-, 2017-04-17:
What other information can I provide?
```
ctest -j 7 --test-load 8
Test project /var/tmp/portage/net-wireless/uhd-3.9.6/work/uhd-3.9.6_build
      Start  1: addr_test
 1/27 Test  #1: addr_test ........................   Passed    0.02 sec
      Start  2: buffer_test
 2/27 Test  #2: buffer_test ......................   Passed    0.03 sec
      Start  3: byteswap_test
 3/27 Test  #3: byteswap_test ....................   Passed    0.01 sec
      Start  4: cast_test
 4/27 Test  #4: cast_test ........................   Passed    0.01 sec
      Start  5: chdr_test
 5/27 Test  #5: chdr_test ........................***Failed    0.01 sec
      Start  6: convert_test
 6/27 Test  #6: convert_test .....................***Failed    0.08 sec
      Start  7: dict_test
 7/27 Test  #7: dict_test ........................   Passed    0.01 sec
      Start  8: error_test
 8/27 Test  #8: error_test .......................   Passed    0.01 sec
      Start  9: fp_compare_delta_test
 9/27 Test  #9: fp_compare_delta_test ............   Passed    0.01 sec
      Start 10: fp_compare_epsilon_test
10/27 Test #10: fp_compare_epsilon_test ..........   Passed    0.01 sec
      Start 11: gain_group_test
11/27 Test #11: gain_group_test ..................   Passed    0.01 sec
      Start 12: math_test
12/27 Test #12: math_test ........................   Passed    0.01 sec
      Start 13: msg_test
13/27 Test #13: msg_test .........................   Passed    0.01 sec
      Start 14: property_test
14/27 Test #14: property_test ....................   Passed    0.01 sec
      Start 15: ranges_test
15/27 Test #15: ranges_test ......................   Passed    0.01 sec
      Start 16: sid_t_test
16/27 Test #16: sid_t_test .......................***Failed    0.01 sec
      Start 17: sph_recv_test
17/27 Test #17: sph_recv_test ....................   Passed    0.02 sec
      Start 18: sph_send_test
18/27 Test #18: sph_send_test ....................   Passed    0.01 sec
      Start 19: subdev_spec_test
19/27 Test #19: subdev_spec_test .................   Passed    0.01 sec
      Start 20: time_spec_test
      Start 21: vrt_test
20/27 Test #21: vrt_test .........................***Failed    0.10 sec
      Start 22: eeprom_c_test
21/27 Test #22: eeprom_c_test ....................***Failed    0.10 sec
      Start 23: error_c_test
22/27 Test #23: error_c_test .....................***Failed    0.10 sec
      Start 24: ranges_c_test
23/27 Test #24: ranges_c_test ....................***Failed    0.10 sec
      Start 25: sensors_c_test
24/27 Test #20: time_spec_test ...................   Passed    0.51 sec
25/27 Test #25: sensors_c_test ...................***Failed    0.01 sec
      Start 26: string_vector_c_test
26/27 Test #26: string_vector_c_test .............***Failed    0.01 sec
      Start 27: subdev_spec_c_test
27/27 Test #27: subdev_spec_c_test ...............***Failed    0.01 sec

63% tests passed, 10 tests failed out of 27

Total Test time (real) =   0.84 sec

The following tests FAILED:
	  5 - chdr_test (Failed)
	  6 - convert_test (Failed)
	 16 - sid_t_test (Failed)
	 21 - vrt_test (Failed)
	 22 - eeprom_c_test (Failed)
	 23 - error_c_test (Failed)
	 24 - ranges_c_test (Failed)
	 25 - sensors_c_test (Failed)
	 26 - string_vector_c_test (Failed)
	 27 - subdev_spec_c_test (Failed)
```

mbr0wn, 2017-04-17:
Can you please post the output of

    $ ctest --output-on-failure

? Also, which platform is this on?

ZeroChaos-, 2017-04-18:
x86_64 platform

requested output is unavailable because, of course, I can't reproduce any longer.

Are there maybe some deps which I accidentally updated?  Is gnuradio perhaps required for these tests and I had an old version?

mbr0wn, 2017-04-19:
GNU Radio is not required. Really, nothing is required other than the build requirements. I also can't immediately see a pattern between the failing tests... I'll close this for now, please reopen should it come back. Also, which OS were you running? If Linux, which distro?

closed 2017-04-19 by mbr0wn

ZeroChaos-, 2017-04-19:
Linux, gentoo

ekagero, 2017-09-29:
Hello, I also got failures when building:

Test project /root/workarea-uhd/uhd/host/build
      Start  1: addr_test
 1/32 Test  #1: addr_test ........................   Passed    0.01 sec
      Start  2: buffer_test
 2/32 Test  #2: buffer_test ......................   Passed    0.03 sec
      Start  3: byteswap_test
 3/32 Test  #3: byteswap_test ....................   Passed    0.01 sec
      Start  4: cast_test
 4/32 Test  #4: cast_test ........................   Passed    0.01 sec
      Start  5: chdr_test
 5/32 Test  #5: chdr_test ........................   Passed    0.01 sec
      Start  6: convert_test
 6/32 Test  #6: convert_test .....................***Failed    0.06 sec
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

Running 20 test cases...
unknown location(0): fatal error: in "test_convert_types_s8_and_s8": uhd::key_error: LookupError: KeyError: Cannot find a conversion routine for conversion ID
  Input format:  s8
  Num inputs:    1
  Output format: s8_item32_le
  Num outputs:   1

/root/workarea-uhd/uhd/host/tests/convert_test.cpp(478): last checkpoint: "test_convert_types_s8_and_s8" entry.
unknown location(0): fatal error: in "test_convert_types_s16_and_s16": uhd::key_error: LookupError: KeyError: Cannot find a conversion routine for conversion ID
  Input format:  s16
  Num inputs:    1
  Output format: s16_item32_le
  Num outputs:   1

/root/workarea-uhd/uhd/host/tests/convert_test.cpp(516): last checkpoint: "test_convert_types_s16_and_s16" entry.
unknown location(0): fatal error: in "test_convert_types_f32_and_f32": uhd::key_error: LookupError: KeyError: Cannot find a conversion routine for conversion ID
  Input format:  f32
  Num inputs:    1
  Output format: f32_item32_le
  Num outputs:   1

/root/workarea-uhd/uhd/host/tests/convert_test.cpp(600): last checkpoint: "test_convert_types_f32_and_f32" entry.

*** 3 failures are detected in the test module "Master Test Suite"

      Start  7: dict_test
 7/32 Test  #7: dict_test ........................   Passed    0.01 sec
      Start  8: error_test
 8/32 Test  #8: error_test .......................   Passed    0.01 sec
      Start  9: fp_compare_delta_test
 9/32 Test  #9: fp_compare_delta_test ............   Passed    0.01 sec
      Start 10: fp_compare_epsilon_test
10/32 Test #10: fp_compare_epsilon_test ..........   Passed    0.01 sec
      Start 11: gain_group_test
11/32 Test #11: gain_group_test ..................   Passed    0.01 sec
      Start 12: math_test
12/32 Test #12: math_test ........................   Passed    0.01 sec
      Start 13: msg_test
13/32 Test #13: msg_test .........................   Passed    0.01 sec
      Start 14: property_test
14/32 Test #14: property_test ....................   Passed    0.01 sec
      Start 15: ranges_test
15/32 Test #15: ranges_test ......................   Passed    0.01 sec
      Start 16: sid_t_test
16/32 Test #16: sid_t_test .......................   Passed    0.01 sec
      Start 17: sph_recv_test
17/32 Test #17: sph_recv_test ....................   Passed    0.01 sec
      Start 18: sph_send_test
18/32 Test #18: sph_send_test ....................   Passed    0.01 sec
      Start 19: subdev_spec_test
19/32 Test #19: subdev_spec_test .................   Passed    0.01 sec
      Start 20: time_spec_test
20/32 Test #20: time_spec_test ...................   Passed    0.51 sec
[... 36 lines cut ...]

Two false, AND: 
nocscript_expr_test: symbol lookup error: nocscript_expr_test: undefined symbol: _ZN3uhd12syntax_errorC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

      Start 31: nocscript_ftable_test
31/32 Test #31: nocscript_ftable_test ............   Passed    0.01 sec
      Start 32: nocscript_parser_test
32/32 Test #32: nocscript_parser_test ............***Failed    0.01 sec
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

nocscript_parser_test: symbol lookup error: nocscript_parser_test: undefined symbol: _ZN3uhd12syntax_errorC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

84% tests passed, 5 tests failed out of 32

Total Test time (real) =   0.85 sec

The following tests FAILED:
	  6 - convert_test (Failed)
	 22 - expert_test (Failed)
	 23 - fe_conn_test (Failed)
	 30 - nocscript_expr_test (Failed)
	 32 - nocscript_parser_test (Failed)
Errors while running CTest

Platform is Kali Rolling. Hope somebody can help

mbr0wn, 2017-10-02:
@ekagero Does a more recent version of UHD work?