EttusResearch/uhd #99: Compiling problem

Y2Kot, 2017-04-21:
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - found
-- 
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 0
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = TRUE
CMake Error at cmake/Modules/UHDComponent.cmake:51 (MESSAGE):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:368 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/uhd/host/build/CMakeFiles/CMakeOutput.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.

This problem takes place on Ubuntu 14.04, on Ubuntu 16.04 I've installed without any problems

noc0lour, 2017-04-21:
Apparently Boost is missing on your Ubuntu 14.04 installation.

Y2Kot, 2017-04-24:
I've installed it

mbr0wn, 2017-04-24:
OK, until we're sure this is actually a bug in UHD I'm closing this. Please take the discussion to usrp-users, and we can help you there much better.

brchiu, 2017-04-25:
hi @Y2Kot, please also check corresponding package version on your Ubuntu 14.04 system. 

For example, if it results from libboost or libboost's development file, then we can see the following version requirement in `hosts/CMakeLists.txt`.
```
SET(Boost_ADDITIONAL_VERSIONS
    "1.46.0" "1.46" "1.47.0" "1.47" "1.48.0" "1.48" "1.48.0" "1.49" "1.50.0" "1.50"
    "1.51.0" "1.51" "1.52.0" "1.52" "1.53.0" "1.53" "1.54.0" "1.54" "1.55.0" "1.55"
    "1.56.0" "1.56" "1.57" "1.57" "1.58" "1.59" "1.60" "1.61"
)
FIND_PACKAGE(Boost 1.53 COMPONENTS ${BOOST_REQUIRED_COMPONENTS})
```

By the way, perhaps you can paste your `/usr/local/src/uhd/host/build/CMakeFiles/CMakeOutput.log` on gist for other people checking.

Y2Kot, 2017-04-28:
Here it is: https://pastebin.com/Cf5f3raA

Y2Kot, 2017-04-28:
Ok, I googled a bit more and find they key - if I install liboost using "sudo apt-get install libboost-all-dev" it's work well. I got this problem on Ubuntu 14.04LTS, on 16.04 I i didn't get problems. Hope this message will be usefull for anyone.

closed 2017-04-28 by Y2Kot