EttusResearch/uhd #102: libUHD aborts when a single UDP control packet is lost

awalls-cx18, 2017-06-17:
libUHD aborts when a single UDP control packet is lost with the following message:

`terminate called after throwing an instance of 'uhd::io_error'`
 ` what():  EnvironmentError: IOError: 0/Radio_0 user_reg_read64() failed: EnvironmentError: IOError:` `[0/Radio_0] sr_read64() failed: EnvironmentError: IOError: Block ctrl (CE_01_Port_40) no response` `packet - AssertionError: bool(buff) `
`  in uint64_t ctrl_iface_impl::wait_for_ack(bool) `
`  at /home/rhino/target/src/uhd/host/lib/rfnoc/ctrl_iface.cpp:197`
`
Aborted`

This problem is observed with an X310, CentOS 7, and UHD master from this commit:

`commit 2790b51f3072b5ad5e3a5c4921a84f460096f92d `
`Merge: 068de67 fae9468`
`Author: Martin Braun <martin.braun@ettus.com>`
`Date:   Thu Apr 13 15:33:03 2017 -0700`

`    Merge branch 'maint'`

The NIC is a Mellonix 40 GB Ethernet card, but only using 10 GB to the X310 obviously.
The Rx sample rate is 250 ksps.
The Tx sample rate is 500 ksps.
Intermittent timed bursts are being transmitted (tx_sob, tx_time, tx_eob).
The network interface MTU was set to 8192

The Network layer was tweaked as follows:

`# Ensure the powersaving driver that might be in use is set`
`# to performance.`
`/bin/cpupower frequency-set -g performance`

`# Tweak network stack and network driver queuing.`
`# FIXME need to look at setting up a queue-discipline`
`# as well.`

`# Set tx queuing discipline default tx queue length to something short`
`/sbin/ip link set ethnet1 txqlen 5`

`# Set NIC driver queue depth low for tx, high for rx`
`/sbin/ethtool -G ethnet1 tx 512`
`/sbin/ethtool -G ethnet1 rx 4096`

`# Disable various segmentation offload functions to have`
`# more deterministic tx latency`
`/sbin/ethtool -K ethnet1 tso off ufo off gso off gro off`

`# Dynamic byte queue limits: Lowering limit_max (default is 1879048192)`
`for i in /sys/class/net/ethnet1/queues/tx-*/byte_queue_limits/limit_max; do`
`    echo 10000 > $i`
`done`
`for i in /sys/class/net/ethnet1/queues/tx-*/byte_queue_limits/limit_min; do`
`    echo 256 > $i`
`done`

An overview of Linux network stack queueing and NIC queuing is here:
[https://www.coverfire.com/articles/queueing-in-the-linux-network-stack/](url)

Linux Byte Queue Limits are explained here:
[https://www.linuxplumbersconf.org/2012/wp-content/uploads/2012/08/bql_slide.pdf](url)

If the byte queue limits are too large, all my timed bursts get a bunch of 'L's (late) emitted by libUHD.
If the byte queue limits are too small, all my timed bursts has 1 to 3 'S's (sequence error) emitted by libUHD.

When I have the BQLs set to something that sees to work.  Things work for a few minutes and the UHD terminates my application with this uhd io error, likely because a control UDP packet was lost.
Is there any way to handle this control/ack packet loss in libUHD without it aborting and killing the application?

awalls-cx18, 2017-06-17:
As a follow up to this, it seems the X310 becomes no-responsive in some respect.  uhd_usrp_probe after the failure yields:

`[INFO] [UHD] linux; GNU C++ version 4.8.5 20150623 (Red Hat 4.8.5-11); Boost_105500; UHD_3.11.0.<blah_blah>`
`[INFO] [X300] X300 initialization sequence...`
`[INFO] [X300] Determining maximum frame size... `
`[INFO] [X300] Maximum frame size: 8000 bytes.`
`[INFO] [X300] Setup basic communication...`
`[INFO] [X300] Loading values from EEPROM...`
`[INFO] [X300] Setup RF frontend clocking...`
`[INFO] [X300] Radio 1x clock:200`
`[INFO] [X300] Detecting internal GPSDO.... `
`[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a`
`[WARNING] [GPS] update_cache: Short GPSDO string: `
`[ERROR] [UHD] Exception caught in safe-call.`
`  in virtual ctrl_iface_impl::~ctrl_iface_impl()`
`  at /home/rhino/target/src/uhd/host/lib/rfnoc/ctrl_iface.cpp:76`
`this->peek32(0); -> EnvironmentError: IOError: Block ctrl (CE_00_Port_30) no response packet -` `AssertionError: bool(buff)`
`  in uint64_t ctrl_iface_impl::wait_for_ack(bool)`
`  at /home/rhino/target/src/uhd/host/lib/rfnoc/ctrl_iface.cpp:197`

`Error: EnvironmentError: IOError: Block ctrl (CE_00_Port_30) no response packet - AssertionError: ` `bool(buff)`
`  in uint64_t ctrl_iface_impl::wait_for_ack(bool)`
`  at /home/rhino/target/src/uhd/host/lib/rfnoc/ctrl_iface.cpp:197`

and power needs to be cycled on the X310 before it responds again.  So maybe this is a firmware thing?

`uhd_usrp_probe --string /mboards/0/fpga_version`
reports `33.0`

`uhd_usrp_probe --string /mboards/0/fpga_version_hash`
reports `000000f`

`uhd_usrp_probe --string /mboards/0/fw_version`
reports `5.0`

awalls-cx18, 2017-06-18:
I rolled libUHD back to the UHD-9.3-LTS branch (v9.3.6 with some fixes) and timed Tx bursts work without a hitch now, along with the ATR GPIOs, which I'm using to key an external amplifier.

I'll wildly speculate that something with the conversion of the X310 base FPGA image to RFNoC broke timed Tx bursts and/or ATR GPIOs.

I forgot to mention, that aside from the Tx and Rx streamers, I have a separate thread querying for GPS NMEA strings every 2 seconds.  In case it matters.

mbr0wn, 2017-06-21:
Hm, interesting. I would not have expected 3.9 to work differently than 3.10 in this respect. In no case do we handle loss of control packets very well (for data packets, we're a bit more lenient).

If it's not actually packet loss that's causing this (and your experiment with 3.9 would confirm this), then we'll have another bug to look into. Timed Tx; we'll have to re-test this.

awalls-cx18, 2017-06-22:
An additional observation:

UHD 3.9.6 seems to have noticeably lower lantency when processing timed Tx bursts vs UHD-3.10.x.
I'm implementing a TDMA system with timeslots that are 26.67 ms long.
With 3.9.6, I can send over timed TX packets with a lead time of 4 timeslots (106.7ms) and never get a late packet gripe from libUHD.
With 3.10.x, I will always get late packet gripes from libUHD with a lead time of only 4 timelsots (106.7ms).  When I use a lead time of 9 timeslots (240 ms), most of the timed Tx packets are OK, with an infrequent late packet.
(Those lead times of 4 and 9 timeslots are not thresholds between working and non-working, they are just times I happened to pick for testing.)

zz4fap, 2018-11-07:
@mbr0wn was there any progress on this issue? Has it been corrected? if so, which is the version it was pushed to? Thanks!

awalls-cx18, 2018-11-07:
I can report that my problem with lost packets, between the host and the X310, was resolved by replacing the cheap SFP adapters, I had been given, with high quality SFP adapters from a reputable manufacturer.

ijgrivell, 2018-11-09:
I am also experiencing this issue, with an X310 and UHD 3.13.0.1. I have tried using a different brand SFP adapter, but this did not resolve the problem.

natetemple, 2019-07-26:
@ijgrivell Is this still an outstanding issue for you? Can you please try using the latest stable version of UHD, 3.14.1.0 and report if it resolves your issue?

natetemple, 2019-08-12:
@ijgrivell I'm closing out this issue, if it is still an outstanding problem for you, please reopen the issue or email us at support@ettus.com.

closed 2019-08-12 by natetemple