EttusResearch/uhd #103: building uhd fpga on windows using vivado and cygwin occurs errors

jackwangRabbit, 2017-06-21:
I'm a new guy who interests in gnu radio. Recently I'm  building uhd fpga(X300) on windows using vivado(15.4) and cygwin(newest), there are always some errors。I used several versions including the maint and released. The errors seem the same.  It is very thankful if I can get some help. Thank you.

below is the log:
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository 'F:/Xilinx/Vivado/2015.4/dat                                                                                                    a/ip'.
BUILDER: Adding XDC     : F:\fg\usrp3\top\x300\x300_1ge.xdc
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\ip\one_gig_eth_pcs_pma\one_gige_p                                                                                                    hy_clk_gen.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\ip\one_gig_eth_pcs_pma\one_gige_p                                                                                                    hy.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\one_gig_eth_pcs_pma_example_design.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\one_gig_eth_pcs_pma_reset_sync_ex.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\one_gig_eth_pcs_pma_sync_block_ex.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\one_gig_eth_pcs_pma_tx_elastic_buffer.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\support\one_gig_eth_pcs_pma_clocking.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\support\one_gig_eth_pcs_pma_gt_common.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\support\one_gig_eth_pcs_pma_resets.v
BUILDER: Adding Verilog : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma_example\one_gig_eth_pcs_pma_example.srcs\sources_1\imports\example_                                                                                                    design\support\one_gig_eth_pcs_pma_support.v
BUILDER: Adding IP      : F:\fg\usrp3\top\x300\build-ip\xc7k325tffg900-2\one_gig                                                                                                    _eth_pcs_pma\one_gig_eth_pcs_pma.xci
BUILDER: Adding XDC     : F:\fg\usrp3\top\x300\x300_10ge_port1.xdc
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\LvFpga_Chinch_Interface.v
BUILDER: Adding Netlist : F:\fg\usrp3\lib\io_port2\LvFpga_Chinch_Interface.ngc
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\ioport2_msg_codec.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\pcie_pkt_route_specifier.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\pcie_axi_wb_conv.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\pcie_wb_reg_core.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\pcie_iop2_msg_arbiter.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\pcie_basic_regs.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\pcie_dma_ctrl.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\data_swapper_64.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\io_port2\pcie_lossy_samp_gate.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\axis_packet_debug.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\chdr_eth_framer.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\cvita_chunker.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\cvita_dechunker.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\cvita_dest_lookup.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\cvita_insert_tlast.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\cvita_packet_debug.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\eth_dispatch.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\eth_interface.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\fix_short_packet.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\ip_hdr_checksum.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\source_flow_control.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\packet_proc\vita_eth_framer.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\vita\flow_control_responder.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\vita\packet_error_responder.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\acc.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\add2_and_clip_reg.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\add2_and_clip.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\add2_and_round_reg.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\add2_and_round.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\add2_reg.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\add2.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\add_then_mac.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\cic_decim.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\cic_dec_shifter.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\cic_interp.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\cic_int_shifter.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\cic_strober.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\clip_reg.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\clip.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\cordic_stage.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\cordic_z24.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\ddc_chain.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\duc_chain.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\hb47_int.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\hb_dec.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\hb_interp.v
BUILDER: [WARNING] File ignored!!!: F:\fg\usrp3\lib\dsp\Makefile.srcs
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\round_reg.v
BUILDER: Adding Verilog : F:\fg\usrp3\lib\dsp\round_sd.v
[... 15 lines cut ...]

read_ip: Time (s): cpu = 00:00:02 ; elapsed = 00:00:10 . Memory (MB): peak = 301                                                                                                    .180 ; gain = 49.266
ERROR: [Common 17-39] 'read_ip' failed due to earlier errors.

    while executing
"read_ip $src_file"
    invoked from within
"if [expr [lsearch {.vhd .vhdl} $src_ext] >= 0] {
            puts "BUILDER: Adding VHDL    : $src_file"
            read_vhdl -library work $src_file
..."
    ("foreach" body line 3)
[... cut ...]
        set src_ext [file extension $src_file ]
        if [expr [lsearch {.vhd .vhdl} $src_ext] >= 0] {
          ..."
    (procedure "vivado_utils::initialize_project" line 19)
    invoked from within
"vivado_utils::initialize_project"
    (file "F:\fg\usrp3\top\x300\build_x300.tcl" line 9)
INFO: [Common 17-206] Exiting Vivado at Tue Jun 20 11:19:46 2017...
make[1]: *** [Makefile.x300.inc:111：bin] error 1
make[1]: leaving “/cygdrive/f/fg/usrp3/top/x300”
make: *** [Makefile:65：X300_HG] error 2

mdmikh, 2017-09-28:
I hit the same ddr3_32bit build error on Windows:

 [IP_Flow 19-3475] Tcl error in ::ipgui_ddr3_32bit::updateAllModelParams procedure for IP 'ddr3_32bit'. Loading device for application Rf_Device from file '7k410t.nph' in environment
C:/Xilinx/Vivado/2015.4/ids_lite/ISE.
child killed: floating-point exception

Workaround I found is the following:

`$ cd build-ip/xc7k410tffg900-2/ddr3_32bit/`
`$ rm mig_a.prj`
`$ cp mig_xc7k410tffg900-2.prj mig_a.prj`

It looks like tcl does not like mig_a.prj being a link on Windows.

michael-west, 2018-04-23:
Fixed as of UHD 3.11.0.0.  (See https://github.com/EttusResearch/fpga/commit/6b33584dced9cbd0166afed231d1913ba69abcc6#diff-c35c9c8e1aee2976e6cc88c7f9600e4d)

closed 2018-04-23 by michael-west