EttusResearch/uhd #109: uhd_images_downloader just deleted my home directory

awelkie, 2017-07-12:
I wanted to get the firmware image, but I didn't want to run uhd_images_downloader.py as root, so I tried to install to my home directory:

```
$ cd
$ uhd_images_downloader -i .
```

But apparently this deletes everything in the specified directory, _recursively_, by default, and without asking the user first? What. The. Fuck. Now my home directory is gone. This can't be expected behavior.

dkozel, 2017-07-12:
@awelkie Clearly that is a horrible outcome.

We're working on a solution to making this very difficult for someone to accidentally do this in the future and will get that onto all active branches very shortly.

mbr0wn, 2017-07-12:
@awelkie This is very unfortunate. I hope no major data losses were incurred. The intention of `-i` was to point it to an existing images directory, which would then get nuked. As @dkozel says, we'll make it hard to accidentally remove things ASAP.

We're tracking this internally and will have a fix out very soon, including on the 3.9.7 release candidate. I'll close this issue.

closed 2017-07-12 by mbr0wn

neelpandeya, 2017-07-12:
@awelkie I would like to talk to you further about this issue. Could you please contact me directly at `neel.pandeya@ettus.com`? Thank you.

dkozel, 2017-07-31:
Changes were introduced on July 17 in https://github.com/EttusResearch/uhd/commit/684fd3a3ffaf31b9fa726ccf2fdea2627f6ea10d which make the default choices safer when specifying custom directories.