EttusResearch/uhd #110: C API: making sensor value handles

n-west, 2017-07-19:
The C API for sensor values exposes `uhd_sensor_value_handle` which is used to get sensor values. All other handles have a generic `_make` function that accepts a pointer to the handle. uhd_sensor_value_handle has uhd_sensor_value_make_from_<datatype> that takes a sensor name, value, and units which is awkward if you just want to create the handle to read values. The underlying class has a `sensor_value_t (const sensor_value_t &source)` ctor that could be exposed to allow a slightly more natural `uhd_sensor_value_make(*handle)` or `uhd_sensor_value_make_from_value(*handle, sensor_value_t *value)`

mbr0wn, 2017-07-20:
@n-west Don't `uhd_sensor_value_make_from_realnum()`, `uhd_sensor_value_make_from_string()` do this? I agree there's no API call to generate an empty sensor value, which we could add (although I'm not sure what you need it for). But it seems the second use case you asked for is already covered. Or I'm misunderstanding.

n-west, 2017-07-20:
It's syntactic sugar. See https://github.com/pysdr/pysdruhd/blob/master/usrp_wrapper.h#L468 for the use case. I just want to set up a return value for reading from the sensor. I don't know what type the sensor will return, so I have to make one up and initialize it. Much tastier to just have uhd_sensor_value_make(*handle)

mbr0wn, 2017-07-20:
Yup, I got it. Makes sense. Will add.

mbr0wn, 2017-07-20:
@n-west Like this?

https://github.com/EttusResearch/uhd/commit/dcca7a4b4e65632c413113809921c3f0aaffc0c4

mbr0wn, 2017-07-20:
OK, it needs to go through internal CI and review before I merge and push public. I'll close this issue now though. Expect it go public by not later than Tuesday.

closed 2017-07-20 by mbr0wn