EttusResearch/uhd #111: Cannot build Python API for python 3

awelkie, 2017-07-19:
This is what happens when I try to build the python API for python 3:

```
$ cmake -DENABLE_PYTHON3=ON ..
-- The CXX compiler identification is GNU 7.1.1
-- The C compiler identification is GNU 7.1.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
--
-- Configuring the python interpreter...
-- Found PythonInterp: /usr/bin/python (found version "2.7.13")
-- Python interpreter: /usr/bin/python
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Working off of feature or development branch. Updating version number.
-- Using UHD Images Directory: OFF
-- Build type not specified: defaulting to release.
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN - Success
--
-- Configuring Boost C++ Libraries...
-- Looking for optional Boost components...
CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1536 (message):
  No header defined for python3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:315 (FIND_PACKAGE)

-- Could NOT find Boost
-- Looking for required Boost components...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Boost version: 1.63.0
-- Found the following Boost libraries:
--   chrono
--   date_time
--   filesystem
--   program_options
--   regex
--   system
--   unit_test_framework
--   serialization
--   thread
--   atomic
-- Boost include directories: /usr/include
-- Boost library directories: /usr/lib64
-- Boost libraries: /usr/lib64/libboost_chrono.so;/usr/lib64/libboost_date_time.so;/usr/lib64/libboost_filesystem.so;/usr/lib64/libboost_program_options.so;/usr/lib64/libboost_regex.so;/usr/lib64/libboost_system.so;/usr/lib64/libboost_unit_test_framework.so;/usr/lib64/libboost_serialization.so;/usr/lib64/libboost_thread.so;/usr/lib64/libboost_atomic.so
-- Looking for log2
-- Looking for log2 - found
--
-- Python checking for Python version 3 or greater
-- Python checking for Python version 3 or greater - "assert platform.python_version() >= '3.0'" failed
--
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - found
--
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - found
--
-- Python checking for numpy 1.7 or greater
-- Python checking for numpy 1.7 or greater - found
-- Could NOT find PythonInterp: Found unsuitable version "2.7.13", but required is at least "3" (found /usr/bin/python)
-- Could NOT find PythonLibs: Found unsuitable version "2.7.13", but required is at least "3" (found /usr/lib64/libpython2.7.so)
--
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = FALSE
--   Dependency HAVE_PYTHON_MODULE_MAKO = TRUE
CMake Error at cmake/Modules/UHDComponent.cmake:51 (MESSAGE):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:416 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
```

This is on Fedora 26. I have both python2 and python3 installed, but the default `python` binary is python2.

awelkie, 2017-07-19:
For some reason I had to point cmake to my python3 binary:

```
$ cmake -DENABLE_PYTHON3=ON -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3 ..
```

Based on Paul's answer to [this question](https://stackoverflow.com/questions/15291500/i-have-2-versions-of-python-installed-but-cmake-is-using-older-version-how-do)

This solved my build issues. I'll leave this issue open since it seems like it may be a problem in the CMakeLists.txt file.

mbr0wn, 2018-04-26:
Merged #163 into python-api to fix this.

closed 2018-04-26 by mbr0wn