EttusResearch/uhd #113: Build fails with "undefined reference to..." in ../lib/libuhd.so

mjpauly, 2017-07-24:
My goal is to build the python API for python 2.7. This is what happens:

```
[ 60%] Linking C executable tx_samples_c
../lib/libuhd.so.3.11: undefined reference to `std::invalid_argument::invalid_argument(std::__cxx11::basic_string<char, std::char
_traits<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::a
ppend(char const*, unsigned long)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `VTT for std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::alloca
tor<char> >@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `boost::match_results<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic
_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char
 const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > >::maybe_assign(boost::match_resul
ts<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, s
td::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>,
 std::allocator<char> > > > > > const&)'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::r
find(char const*, unsigned long, unsigned long) const@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::thread::_State::~_State()@GLIBCXX_3.4.22'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_
M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::s
ubstr(unsigned long, unsigned long) const@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::r
eserve(unsigned long)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::f
ind_last_of(char const*, unsigned long, unsigned long) const@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<cha
r> >::basic_ostringstream(std::_Ios_Openmode)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::length_error::length_error(char const*)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::runtime_error::runtime_error(char const*)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::b
asic_string(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long, unsigned long)
@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::r
esize(unsigned long, char)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `vtable for std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::al
locator<char> >@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_
M_erase(unsigned long, unsigned long)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::logic_error::logic_error(char const*)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::runtime_error::runtime_error(std::__cxx11::basic_string<char, std::char_trait
s<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `typeinfo for std::ios_base::failure[abi:cxx11]@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::numpunct<char> const& std::use_facet<std::__cxx11::numpunct<char> >(
std::locale const&)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail::perl_matcher<__gnu_cxx::__normal_iterator<char const*, std::__cx
x11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_ite
rator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, boost::regex_traits<cha
r, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<c
har> > > const&, boost::regex_constants::_match_flags)'
../lib/libuhd.so.3.11: undefined reference to `std::__throw_out_of_range_fmt(char const*, ...)@GLIBCXX_3.4.20'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::p
ush_back(char)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_
M_append(char const*, unsigned long)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::c
ompare(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `typeinfo for std::thread::_State@GLIBCXX_3.4.22'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::f
ind(char, unsigned long) const@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char
> >::~basic_stringstream()@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::runtime_error::runtime_error(char const*)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_
M_construct(unsigned long, char)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_del
ete<std::thread::_State> >, void (*)())@GLIBCXX_3.4.22'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::a
ssign(char const*)@GLIBCXX_3.4.21'
[... 45 lines cut ...]
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_istringstream<char, std::char_traits<char>, std::allocator<cha
r> >::~basic_istringstream()@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `VTT for std::__cxx11::basic_istringstream<char, std::char_traits<char>, std::alloc
ator<char> >@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::o
perator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_
M_mutate(unsigned long, unsigned long, char const*, unsigned long)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~
basic_string()@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_
M_replace(unsigned long, unsigned long, char const*, unsigned long)@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char
 const*, char const*) const'
../lib/libuhd.so.3.11: undefined reference to `boost::archive::text_oarchive_impl<boost::archive::text_oarchive>::save(std::__cxx
11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../lib/libuhd.so.3.11: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::f
ind(char const*, unsigned long, unsigned long) const@GLIBCXX_3.4.21'
../lib/libuhd.so.3.11: undefined reference to `std::runtime_error::runtime_error(std::__cxx11::basic_string<char, std::char_trait
s<char>, std::allocator<char> > const&)@GLIBCXX_3.4.21'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/tx_samples_c] Error 1
make[1]: *** [examples/CMakeFiles/tx_samples_c.dir/all] Error 2
make: *** [all] Error 2
```

mjpauly, 2017-07-24:
For reference, here's the configuration output:
```
-- The CXX compiler identification is GNU 6.3.0
-- The C compiler identification is GNU 6.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
--
-- Configuring the python interpreter...
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
-- Python interpreter: /usr/bin/python
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Working off of feature or development branch. Updating version number.
-- Using UHD Images Directory: OFF
-- Build type not specified: defaulting to release.
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN - Success
--
-- Configuring Boost C++ Libraries...
-- Looking for optional Boost components...
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   python
-- Looking for required Boost components...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   chrono
--   date_time
--   filesystem
--   program_options
--   regex
--   system
--   unit_test_framework
--   serialization
--   thread
--   atomic
-- Boost include directories: /usr/include
-- Boost library directories: /usr/lib/x86_64-linux-gnu
-- Boost libraries:
/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_f
ilesystem.so;/usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-li
nux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so;/usr/lib/x86_64-linux-gnu/libboost_serializa
tion.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so
-- Looking for log2
-- Looking for log2 - found
--
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
--
-- Python checking for Mako templates 0.4.2 or greater
[... 217 lines cut ...]
--   * N230
--   * OctoClock
--   * Manual
--   * API/Doxygen
--   * Man Pages
--
-- ######################################################
-- # UHD disabled components
-- ######################################################
--   * GPSD
--   * E100
--   * E300
--
-- ******************************************************
-- * You are building a development branch of UHD.
-- * These branches are designed to provide early access
-- * to UHD and USRP features, but should be considered
-- * unstable and/or experimental!
-- ******************************************************
-- Building version: 3.11.0.python-api-242-g73e6726a
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: 
```

brentstapleton, 2017-07-24:
This looks like a linker error, and may be a legitimate issue, but should be unrelated to the python-api (tx_samples_c is part of the C API).

Have you tried starting with a clean build/ directory? I haven't been able to recreate this issue (although I do have slightly different dependency versions).

mjpauly, 2017-07-24:
@brentstapleton Thanks for the reply! I just found the solution. I downgraded gcc back to 4.8.4 and it built successfully. You're right, the issue was unrelated to the python-api.

closed 2017-07-24 by mjpauly