EttusResearch/uhd #119: make test fails on Debian Stretch

oogieoogieful, 2017-08-15:
Hi,
I have python 2.7 (through Anaconda) on Debian Stretch. I have already installed uhd (and their dependencies) with sudo apt-get, but I would like to use the python-api. 

I've cloned the uhd repo locally.  

$ git clone https://github.com/EttusResearch/uhd.git

Navigated to the host folder.  Create the build directory and called:

$ cmake ../

No error.  Then,

$ make

No error.

But when I run:

$ make test

I don't pass all of the tests:

Running tests...
Test project /home/erin/Desktop/UHD/uhd/host/build
      Start  1: addr_test
 1/32 Test  #1: addr_test ........................   Passed    0.01 sec
      Start  2: buffer_test
 2/32 Test  #2: buffer_test ......................   Passed    0.03 sec
      Start  3: byteswap_test
 3/32 Test  #3: byteswap_test ....................   Passed    0.01 sec
      Start  4: cast_test
 4/32 Test  #4: cast_test ........................   Passed    0.01 sec
      Start  5: chdr_test
 5/32 Test  #5: chdr_test ........................   Passed    0.01 sec
      Start  6: convert_test
 6/32 Test  #6: convert_test .....................***Failed    0.05 sec
      Start  7: dict_test
 7/32 Test  #7: dict_test ........................   Passed    0.01 sec
      Start  8: error_test
 8/32 Test  #8: error_test .......................   Passed    0.01 sec
      Start  9: fp_compare_delta_test
 9/32 Test  #9: fp_compare_delta_test ............   Passed    0.01 sec
      Start 10: fp_compare_epsilon_test
10/32 Test #10: fp_compare_epsilon_test ..........   Passed    0.01 sec
      Start 11: gain_group_test
11/32 Test #11: gain_group_test ..................   Passed    0.01 sec
      Start 12: math_test
12/32 Test #12: math_test ........................   Passed    0.01 sec
      Start 13: msg_test
13/32 Test #13: msg_test .........................   Passed    0.01 sec
      Start 14: property_test
14/32 Test #14: property_test ....................   Passed    0.01 sec
      Start 15: ranges_test
15/32 Test #15: ranges_test ......................   Passed    0.01 sec
      Start 16: sid_t_test
16/32 Test #16: sid_t_test .......................   Passed    0.01 sec
      Start 17: sph_recv_test
17/32 Test #17: sph_recv_test ....................   Passed    0.01 sec
      Start 18: sph_send_test
18/32 Test #18: sph_send_test ....................   Passed    0.01 sec
      Start 19: subdev_spec_test
19/32 Test #19: subdev_spec_test .................   Passed    0.01 sec
      Start 20: time_spec_test
20/32 Test #20: time_spec_test ...................   Passed    0.51 sec
      Start 21: vrt_test
21/32 Test #21: vrt_test .........................   Passed    0.01 sec
      Start 22: expert_test
22/32 Test #22: expert_test ......................***Failed    0.01 sec
      Start 23: fe_conn_test
23/32 Test #23: fe_conn_test .....................***Failed    0.01 sec
      Start 24: eeprom_c_test
24/32 Test #24: eeprom_c_test ....................   Passed    0.01 sec
      Start 25: error_c_test
25/32 Test #25: error_c_test .....................   Passed    0.01 sec
      Start 26: ranges_c_test
26/32 Test #26: ranges_c_test ....................   Passed    0.01 sec
      Start 27: sensors_c_test
27/32 Test #27: sensors_c_test ...................   Passed    0.01 sec
      Start 28: string_vector_c_test
28/32 Test #28: string_vector_c_test .............   Passed    0.01 sec
      Start 29: subdev_spec_c_test
29/32 Test #29: subdev_spec_c_test ...............   Passed    0.01 sec
      Start 30: nocscript_expr_test
30/32 Test #30: nocscript_expr_test ..............***Failed    0.01 sec
      Start 31: nocscript_ftable_test
31/32 Test #31: nocscript_ftable_test ............   Passed    0.01 sec
      Start 32: nocscript_parser_test
32/32 Test #32: nocscript_parser_test ............***Failed    0.01 sec

84% tests passed, 5 tests failed out of 32

Total Test time (real) =   0.80 sec

The following tests FAILED:
	  6 - convert_test (Failed)
	 22 - expert_test (Failed)
	 23 - fe_conn_test (Failed)
	 30 - nocscript_expr_test (Failed)
	 32 - nocscript_parser_test (Failed)
Errors while running CTest
Makefile:149: recipe for target 'test' failed
make: *** [test] Error 8

Can anyone help me resolve this issue?

Thank you.

oogieoogieful, 2017-08-15:
Looking at my clone, I don't see a python folder in host or examples.  How do I get the python-api version of uhd?

brentstapleton, 2017-08-15:
You can check your current branch with ```git branch```; if python-api doesn't show up or have an asterisk next to it, you should run ```git checkout --track origin/python-api```.

Can you run those tests with higher verbosity with ```ctest -V -I 6,22,23,30,32``` and post the results?

oogieoogieful, 2017-08-15:
This appears to be a git mixup on my part,

$ git clone https://github.com/EttusResearch/uhd.git --branch python-api --single-branch

I will run the tests again.

oogieoogieful, 2017-08-15:
Now, when I run the tests (with the specified branch) all of the tests pass.

Thank you @brentstapleton for your time.

brentstapleton, 2017-08-15:
Good to hear. FYI, we also have a mailing list for USRP-related discussion and support at usrp-users@lists.ettus.com.

closed 2017-08-15 by brentstapleton