EttusResearch/uhd #121: Overflow Error with UHD Python - API

oogieoogieful, 2017-08-16:
Hello,

I am running a modified version example included with the python-api branch:

As a benchmark, I am comparing it to our current ctypes library:

oogieoogieful, 2017-08-16:
dang, I hit enter before I was ready.

michael-west, 2018-02-16:
@oogieoogieful The error is indicating a dropped packet (hence, the 'D' printed in the output).  This is usually indicative of a network issue and not generally due to a bug in the UHD code.  The other errors reported in the output (i.e. fifo ctrl timed out looking for acks) also indicate packet loss on the network.  I recommend testing UDP performance on your network.  For further support, please consider posting on the [usrp-users mailing list ](http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com)or contacting [Ettus Support](mailto:support@ettus.com).

closed 2018-02-16 by michael-west

oogieoogieful, 2018-04-17:
When I use the uhd c/c++ utilities to capture data (or the in-house capture method), I have no problem with dropped packets. 

I can run the benchmark tests okay for 25e6/50e6 samples/sec capture. 

On the same network using the python api capture I get dropped packets. Therefore, there seems to be a problem the python api.  I originally tested this python api when it was first released.

michael-west, 2018-04-17:
@oogieoogieful Thanks for the feedback.  We will look into the streaming performance of the Python API.

LOVEANDNINENINE, 2023-04-10:
Try to check if you are using more than one program to call your USRP device?