EttusResearch/uhd #122: "bad lexical cast" thrown when calling `uhd_dboard_eeprom_get_revision`

creinke-sandia, 2017-08-22:
I'm reading the EEPROM for the "gdb", "rx", and "tx" units of the daugterboard in slot "A" using the C API.
`uhd_dboard_eeprom_get_id` and `uhd_dboard_eeprom_get_serial` work, although they don't return anything interesting for "gdb".  `uhd_dboard_eeprom_set_revision`, however, produces an error for all of the units, presumably because string value of the underlying property isn't an integer.

```
Throw location unknown (consider using BOOST_THROW_EXCEPTION)
Dynamic exception type: class boost::exception_detail::clone_impl<struct boost::exception_detail::error_info_injector<class boost::bad_lexical_cast> >
std::exception::what: bad lexical cast: source type value could not be interpreted as target
```

mbr0wn, 2017-08-23:
Which device are you running this on? And which dboard? The revision doesn't have to be set. This could be a legit failure.

mbr0wn, 2017-08-23:
Just saw your comment on the device info. That's an X310 + SBX for future reference.

creinke-sandia, 2017-08-23:
uhd_usrp_probe says the daughterboard is
```
RX Dboard: A
ID: SBX v5 (0x0069)
```

mbr0wn, 2017-08-23:
Thanks for the info!

michael-west, 2018-02-16:
The SBX does not have a GDB (granddaughter board) and the TX and RX EEPROMS do not have a revision populated.  If you use the usrp_burn_db_eeprom utility to read the TX and RX EEPROMs, you can see that.  The bad lexical cast error is getting thrown because the revision field is empty.  Admittedly, the error is not descriptive of the real problem and the handling by the API could be improved.

Moving this to internal tracking for triage.

closed 2018-02-16 by michael-west