EttusResearch/uhd #123: Segmentation fault in USRP source in MacOS

vamsi765, 2017-08-27:
I get the following error [I see boost::task_io_service error as final debug]:

> Segmentation fault: 11

When I create the instance of the usurp_source
`
        self.sdr_source = uhd.usrp_source(
        	",".join(('', args)),
        	uhd.stream_args(
        		cpu_format="fc32",
        		channels=range(1),
        	),
        )
`

This code is autogenerated when I create a USRP source.

The error started after I updated to latest UHD using MacPorts (it did work previously).

Here is the log generated by MacOS:
[usrp_source_error.txt](https://github.com/EttusResearch/uhd/files/1254672/usrp_source_error.txt)

michaelld, 2017-08-28:
Does this error occur every time you init the USRP, or just sometimes?

What does "port installed and active" return?

Have you tried the "uhd-devel" port? Sometimes fixes are in there that aren't yet in the release. It is, plus or minus, a drop-in replacement of "uhd".

vamsi765, 2017-08-31:
It does occur all the time.
After updating to und-devel port resolved the issue.

Thanks!

closed 2017-08-31 by vamsi765

michaelld, 2017-08-31:
You're welcome! I'm glad moving to the devel port did the trick.