EttusResearch/uhd #129: Possibly incorrect effective RX center frequency when using DSP frequency offset

ccsh23, 2017-10-04:
Dear all,

I have found possible bug in setting effective RX center frequency when using DSP frequency offset. Say that I want to use RF frequency of 1455 MHz and DSP offset of -5 MHz on both TX and RX sides. When using these settings in TX freq tuning structure, everything works fine and I can see my signal @1450 MHz, as expected. However on RX side I have to use RF frequency of 1455 MHz and DSP offset of +5 MHz in order to have my baseband RX signal centered around 0 Hz again. I believe this is a mistake and it should be possible with using exactly the same DSP offset as the one used on TX side, not its negative value. But maybe I am missing some DSP sign convention?

More detailed description of a problem (with images etc.) may be found on SoapyUHD github tracker, as initially I suspected that this is SoapyUHD-specific issue:
https://github.com/pothosware/SoapyUHD/issues/10

Thank you in advance for fixing/explaining this.

ion-concepts, 2017-10-06:
This is indeed the design intent. For your example:
You constructed a complex modulated signal around baseband and then this signal was digitally offset by -5MHz so that when it was mixed in the analog domain with an RF LO of 1455MHz the modulated signal was located at 1450MHz.
On reception the signal path is traversed in reverse order, the received signal is mixed with an RF LO of 1450MHz placing your received modulated signal at a center of -5MHz. This signal is then digitally offset by +5MHz to yield a received complex modulated signal centered at baseband.

ccsh23, 2017-10-07:
Okay, just wanted to confirm as the interpretation in SoapySDR is different (you are supposed to use the same offset on both TX and RX side).

closed 2018-02-16 by michael-west