EttusResearch/uhd #130: X300 image updater sometimes fails over Ethernet

cnelson711, 2017-10-05:
TLDR: Would resending a UDP packet on failure be a good solution to increase the robustness of the uhd_image_loader for the x300?

Details:
We have updated firmware on ~30 x300 USRPs using uhd_image_loader, and have noticed that the firmware update over Ethernet fails fairly frequently (maybe 3-5% of the time).

After investigating ./host/lib/usrp/x300/x300_image_loader.cpp, it's not really a surprise that a UDP update fails.

It seems like when the updater is sending in the firmware, it sends the sector, index, and number of bytes (256), and the payload of bytes... that and the fact that UDP packets can arrive out of order makes me think that we should be able to resend a packet if it fails.  I.E. If a UDP packet fails on lines 316 or 320 of /host/lib/usrp/x300/x300_image_loader.cpp, retry X times before exiting out - i.e. repeat the packet loop on lines 287-324.  This should decrease the failure rate.

Or better yet, use an error correcting protocol like TCP (but perhaps this is not implemented in the x300's bootloader).

Any advice?

mbr0wn, 2017-10-05:
Can confirm that the X300 FW does not implement TCP. We will investigate increasing the image burner reliability.

michael-west, 2018-02-16:
Moved to internal issue tracking for triage and scheduling.  Thank you for reporting @cnelson711 !

closed 2018-02-16 by michael-west