EttusResearch/uhd #131: More flexible B210 RF data routing

mrjacobagilbert, 2017-10-06:
The USRP B210 has two RF channels, two DDC chains, and the ability to feed the data from a single antenna into both DDCs built into the FPGA (https://github.com/EttusResearch/fpga/blob/master/usrp3/top/b200/b200_io.v#L345), however there does not appear to be a way to do this with UHD. This would be very powerful for low-power systems to be able to leverage both DDC chains to perform 2-channel channelization on the USRP. For example, easily have 2x 125kHz channels within the 56 MHz RTBW of the 9361 where there is no chance of even a powerful host ingesting that amount of bandwidth and performing channelization without data loss).

I have accomplished this with a fairly simple yet egregious hack in host/lib/usrp/b200/b200_impl.cpp in which I tell the device it's operating in MIMO mode, force the value of a specific conf bit https://github.com/EttusResearch/uhd/blob/maint/host/lib/usrp/b200/b200_impl.cpp#L1156 and bypass all the safety checks but this is really something that would be useful to be properly implemented.

Further, extending this capability to single channel devices (B200 / B200mini) by instantiating additional DDC chains could both increase capability and increase standardization across receive capabilities within the product family.

ion-concepts, 2017-10-06:
You're miss-understanding they way that MIMO bit works in that register, it has nothing to do with the internal 9361 configuration (Which is programmed via SPI). That bit determines if the FPGA interface interprets successive bus cycles from the 9361 as interleaved single channel I & Q date or interleaved dual channel I & Q data. If UHD (and hence the 9361) are configured in MIMO mode and you have forced that bit to put the interface in SISO mode then you are interleaving samples from both radio channels, then sending the duplicate interleaved streams to both DDC's. (The DDC's will also incidentally be running at twice the sample rate you programmed) For that feature to work as intended by design, the 9361 must also be in SISO mode.

michael-west, 2017-10-07:
Thank you for posting the request.  The ability to replicate the data from a single channel into multiple DDC chains is certainly possible to do and would be a nice feature to add to the B2xx series.  Unfortunately, it is not a feature planned for the B2xx series at this time.

closed 2017-10-07 by michael-west

mrjacobagilbert, 2017-10-07:
I just checked and what I actually do is comment out that line so that bit is not set.

I certainly won't claim to understand exactly whats going on, but doing what I described is providing me the ability to extract two independent signals that are 50 MHz apart from a single antenna port. This shouldn't be possible if the 9361 is actually operating in MIMO mode. I definitely have two channels which indicates to me that I have the FPGA-host interface in MIMO and the 9361-FPGA in SISO.

Regardless, as stated this is a terrible hack and not a good ultimate solution. I think a more thought out implementation of this would be an excellent feature.