EttusResearch/uhd #132: why is my usrp X310 not connecting to my computer

robellebor, 2017-10-17:
it used to work ... after sometime i uploaded a new image to it and it stopped connecting to my computer

dkozel, 2017-10-17:
Hello robellebor,

Can you provide more information about your setup and what new image you uploaded? The most common connection issues have to do with networking problems.
http://files.ettus.com/manual/page_usrp_x3x0.html#x3x0_getting_started_connectivity

You can reload the image using the USB JTAG if necessary, this lasts only while powered on and allows you to recover from a failed image upload.
http://files.ettus.com/manual/page_usrp_x3x0.html#x3x0_jtag

I recommend sending a message to the support mailing list, more people will be able to help with a situation like this on there. You can also email support@ettus.com.
http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com

Regards,
Derek

closed 2017-10-17 by dkozel

robellebor, 2017-10-18:
the image i uploaded was uhd-images_003.009.002-release ... i have downloaded a new image uhd-images_003.010.002.000-release , but i cant load it to my usrp coz it its not connecting ... i can ping 192.168.10.2 ... 
when i execute this     uhd_image_loader --args="type=x300,addr=192.168.10.2" --fpga-path="<path_to_images>/usrp_x310_fpga_HG.bit"   it says

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-release
No applicable UHD devices found