EttusResearch/uhd #133: Syscall param points to uninitialised

FarshadKoohifar, 2017-10-22:
Hi,

I am running valgrind on my code that interfaces with UHD, and it looks like UHD is using uninitialized memory in some UDP communication. Here is my relevant code:

```
	tx_args=rc->usrp_address_name;
	ref="internal";
	tx_rate=rc->sampleRate; tx_freq=rc->centralFreq; tx_gain=rc->rf_gain_tx; tx_bw=rc->bandWidth;
	rx_args=rc->usrp_address_name;
	rx_rate=rc->sampleRate;	rx_freq=rc->centralFreq; rx_gain=rc->rf_gain_rx; rx_bw=rc->bandWidth;
	tx_usrp = uhd::usrp::multi_usrp::make(tx_args);
	rx_usrp = uhd::usrp::multi_usrp::make(rx_args);
	tx_usrp->set_clock_source(ref);
	rx_usrp->set_clock_source(ref);
```

Here is the valgrind output:
```
==7383== Syscall param sendmsg(msg.msg_iov[0]) points to uninitialised byte(s)
==7383==    at 0x5321A50: __sendmsg_nocancel (syscall-template.S:84)
==7383==    by 0x5CC9CB6: udp_simple_impl::send(boost::asio::const_buffer const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x59B3A5F: uhd::usrp::usrp3::usrp3_fw_ctrl_iface::discover_devices(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x5BC36D6: uhd::usrp::n230::n230_impl::n230_find(uhd::device_addr_t const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x5A27B3F: boost::detail::function::function_invoker1<std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x5CE288B: uhd::device::make(uhd::device_addr_t const&, uhd::device::device_filter_t, unsigned long) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x579F7CE: uhd::usrp::multi_usrp::make(uhd::device_addr_t const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x418CC8: UhdInterface::obtainStreams() (UhdInterface.cc:21)
==7383==    by 0x4188A7: UhdInterface::UhdInterface(RadioConfig&) (UhdInterface.cc:10)
==7383==    by 0x40810A: main (main.cc:83)
==7383==  Address 0xffeffc37c is on thread 1's stack
==7383==  in frame #2, created by uhd::usrp::usrp3::usrp3_fw_ctrl_iface::discover_devices(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short) (???:)
==7383== 
-- X300 initialization sequence...
-- Determining maximum frame size... 1472 bytes.
-- Setup basic communication...
-- Loading values from EEPROM...
-- Setup RF frontend clocking...
-- Radio 1x clock:200
-- [DMA FIFO] Running BIST for FIFO 0... pass (Throughput: 1303.2MB/s)
-- [DMA FIFO] Running BIST for FIFO 1... pass (Throughput: 1303.1MB/s)
-- [RFNoC Radio] Performing register loopback test... pass
-- [RFNoC Radio] Performing register loopback test... pass
-- [RFNoC Radio] Performing register loopback test... pass
-- [RFNoC Radio] Performing register loopback test... pass
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
==7383== Syscall param sendmsg(msg.msg_iov[0]) points to uninitialised byte(s)
==7383==    at 0x5321A6D: ??? (syscall-template.S:84)
==7383==    by 0x5CC9CB6: udp_simple_impl::send(boost::asio::const_buffer const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x59B3A5F: uhd::usrp::usrp3::usrp3_fw_ctrl_iface::discover_devices(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x5BC36D6: uhd::usrp::n230::n230_impl::n230_find(uhd::device_addr_t const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x5A27B3F: boost::detail::function::function_invoker1<std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> > (*)(uhd::device_addr_t const&), std::vector<uhd::device_addr_t, std::allocator<uhd::device_addr_t> >, uhd::device_addr_t const&>::invoke(boost::detail::function::function_buffer&, uhd::device_addr_t const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x5CE288B: uhd::device::make(uhd::device_addr_t const&, uhd::device::device_filter_t, unsigned long) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x579F7CE: uhd::usrp::multi_usrp::make(uhd::device_addr_t const&) (in /usr/local/lib/libuhd.so.003.010)
==7383==    by 0x418D3F: UhdInterface::obtainStreams() (UhdInterface.cc:22)
==7383==    by 0x4188A7: UhdInterface::UhdInterface(RadioConfig&) (UhdInterface.cc:10)
==7383==    by 0x40810A: main (main.cc:83)
==7383==  Address 0xffeffc37c is on thread 1's stack
==7383==  in frame #2, created by uhd::usrp::usrp3::usrp3_fw_ctrl_iface::discover_devices(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short) (???:)
==7383== 
```
and here is my uhd version:

`linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-3-g122bfae1`

Please advise me on how to resolve this.

Thanks,
Farshad

mbr0wn, 2017-10-23:
Is this causing any issues for you? I can see that `request` isn't getting fully populated, but it's also meant to be a dummy request.

mbr0wn, 2017-10-23:
@FarshadKoohifar can you try this:
```patch
From bb4194f579ce84fb339befed810749c85a1cec04 Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Mon, 23 Oct 2017 15:16:30 -0700
Subject: [PATCH] n230: Properly initialize request before discovery

The request struct passed to send() was not fully initialized, causing
code checkers to choke. This fix doesn't affect functionality (the
values were not used for anything), but removes warnings.
---
 host/lib/usrp/n230/n230_fw_ctrl_iface.cpp | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/host/lib/usrp/n230/n230_fw_ctrl_iface.cpp b/host/lib/usrp/n230/n230_fw_ctrl_iface.cpp
index 18c2c4cf8..61a7d2d65 100644
--- a/host/lib/usrp/n230/n230_fw_ctrl_iface.cpp
+++ b/host/lib/usrp/n230/n230_fw_ctrl_iface.cpp
@@ -23,6 +23,7 @@
 #include <boost/format.hpp>
 #include <boost/asio.hpp> //used for htonl and ntohl
 #include "n230_fw_comm_protocol.h"
+#include <cstring>
 
 namespace uhd { namespace usrp { namespace n230 {
 
@@ -189,6 +190,7 @@ std::vector<std::string> n230_fw_ctrl_iface::discover_devices(
 
     //Send dummy request
     fw_comm_pkt_t request;
+    std::memset(&request, 0, sizeof(request));
     request.id = uhd::htonx<uint32_t>(FW_COMM_GENERATE_ID(product_id));
     request.flags = uhd::htonx<uint32_t>(FW_COMM_FLAGS_ACK|FW_COMM_CMD_ECHO);
     request.sequence = uhd::htonx<uint32_t>(std::rand());
-- 
2.14.2
```

FarshadKoohifar, 2017-10-23:
Thanks for the answer, good to know that the request is dummy. 
I will ignore the valgrind report then!

closed 2017-10-23 by FarshadKoohifar

mbr0wn, 2017-10-23:
Thanks for reporting it though! We might still publish a fix.