EttusResearch/uhd #134: Nullpointer exception when calling uhd_tx_streamer_send

FabianEckermann, 2017-10-24:
When using a URSP B210 via USB (srslte) we infrequently get the following error message:

    T* boost::intrusive_ptr<T>::operator->() const [with T = uhd::transport::managed_send_buffer]: Assertion `px != 0' failed.

We're guessing this could be related to an unhandled timeout related to [libusb1_zero_copy](https://github.com/EttusResearch/uhd/blob/master/host/lib/transport/libusb1_zero_copy.cpp#L147-L159), but we couldn't verify it yet.

mbr0wn, 2017-10-24:
Interesting -- if you can point us to a way to reproduce this, that would certainly be very helpful. Thanks for reporting!

FabianEckermann, 2017-10-25:
We use the latest version of [srsLTE](https://github.com/srsLTE/srsLTE) with an USRP B210.
When starting the srsenb or srsue the error infrequently occurs. A precise reproduction is not possible from what we inspected so far.

michael-west, 2018-02-16:
@FabianEckermann I tried to reproduce the issue by repeatedly starting srsenb with the example configuration, but I was unable to reproduce it.

For us to be able to troubleshoot any further, we need a bit more information.  Please provide the output of uhd_config_info --print-all and a stack trace of the failure as well as steps needed to reproduce the failure.  Having your srsenb configuration files and host system specifications (especially the USB Host Controller information) might also be helpful.

I don't suspect an unhandled timeout, but it could be an uhandled USB error that is only seen on some systems in very certain circumstances.

michael-west, 2018-03-27:
We are closing this because we were unable to reproduce the issue and no additional information has been provided.  If the issue persists, please re-open the issue and provide more details on how to reproduce it.  Thank you.

closed 2018-03-27 by michael-west

michael-west, 2018-04-26:
Root cause identified in #171.