EttusResearch/uhd #136: uhd compilation issue

pikumar, 2017-10-30:
On a VM when I try to compile UHD I get the following error:
(Any chance one can avoid this bug/error?)

Reading package lists... Done
Building dependency tree
Reading state information... Done
autoconf is already the newest version (2.69-9).
automake is already the newest version (1:1.15-4ubuntu1).
build-essential is already the newest version (12.1ubuntu2).
ccache is already the newest version (3.2.4-1).
doxygen is already the newest version (1.8.11-1).
g++ is already the newest version (4:5.3.1-1ubuntu1).
libasound2-dev is already the newest version (1.1.0-0ubuntu1).
libfftw3-bin is already the newest version (3.3.4-2ubuntu1).
libfftw3-dev is already the newest version (3.3.4-2ubuntu1).
libfftw3-doc is already the newest version (3.3.4-2ubuntu1).
libgsl-dev is already the newest version (2.1+dfsg-2).
libncurses5 is already the newest version (6.0+20160213-1ubuntu1).
libncurses5-dbg is already the newest version (6.0+20160213-1ubuntu1).
libncurses5-dev is already the newest version (6.0+20160213-1ubuntu1).
liborc-0.4-0 is already the newest version (1:0.4.25-1).
liborc-0.4-dev is already the newest version (1:0.4.25-1).
libqt4-dev is already the newest version (4:4.8.7+dfsg-5ubuntu2).
libqt4-dev-bin is already the newest version (4:4.8.7+dfsg-5ubuntu2).
libsdl1.2-dev is already the newest version (1.2.15+dfsg1-3).
libtool is already the newest version (2.4.6-0.1).
libusb-1.0-0 is already the newest version (2:1.0.20-1).
libusb-1.0-0-dev is already the newest version (2:1.0.20-1).
libusb-dev is already the newest version (2:0.1.12-28).
libxi-dev is already the newest version (2:1.7.6-1).
libxrender-dev is already the newest version (1:0.9.9-0ubuntu1).
ncurses-bin is already the newest version (6.0+20160213-1ubuntu1).
python-dev is already the newest version (2.7.11-1).
python-docutils is already the newest version (0.12+dfsg-1).
python-gtk2 is already the newest version (2.24.0-4ubuntu1).
python-lxml is already the newest version (3.5.0-1build1).
python-mako is already the newest version (1.0.3+ds1-1ubuntu1).
python-numpy is already the newest version (1:1.11.0-1ubuntu1).
python-numpy-dbg is already the newest version (1:1.11.0-1ubuntu1).
python-numpy-doc is already the newest version (1:1.11.0-1ubuntu1).
python-requests is already the newest version (2.9.1-3).
python-tk is already the newest version (2.7.11-2).
qt4-bin-dbg is already the newest version (4:4.8.7+dfsg-5ubuntu2).
qt4-dev-tools is already the newest version (4:4.8.7+dfsg-5ubuntu2).
qt4-doc is already the newest version (4:4.8.7+dfsg-5ubuntu2).
cpufrequtils is already the newest version (008-1).
fort77 is already the newest version (1.15-10).
libboost-all-dev is already the newest version (1.58.0.1ubuntu1).
libcomedi-dev is already the newest version (0.10.2-4).
libcppunit-1.13-0v5 is already the newest version (1.13.2-2.1).
libcppunit-dev is already the newest version (1.13.2-2.1).
libcppunit-doc is already the newest version (1.13.2-2.1).
libqwt5-qt4-dev is already the newest version (5.2.3-1).
libqwt6abi1 is already the newest version (6.1.2-5).
libqwtplot3d-qt4-dev is already the newest version (0.2.7+svn191-10).
libzmq-dev is already the newest version (2.2.0+dfsg-7).
libzmq1 is already the newest version (2.2.0+dfsg-7).
pyqt4-dev-tools is already the newest version (4.11.4+dfsg-1build4).
python-cheetah is already the newest version (2.4.4-3.fakesyncbuild1).
python-opengl is already the newest version (3.0.2-1).
python-qt4 is already the newest version (4.11.4+dfsg-1build4).
python-qt4-dbg is already the newest version (4.11.4+dfsg-1build4).
python-qt4-dev is already the newest version (4.11.4+dfsg-1build4).
python-qt4-doc is already the newest version (4.11.4+dfsg-1build4).
python-qwt5-qt4 is already the newest version (5.2.1~cvs20091107+dfsg-7build1).
python-scipy is already the newest version (0.17.0-1).
python-wxgtk3.0 is already the newest version (3.0.2.0+dfsg-1build1).
python-zmq is already the newest version (15.2.0-0ubuntu4).
qt4-default is already the newest version (4:4.8.7+dfsg-5ubuntu2).
r-base-dev is already the newest version (3.2.3-4).
[... 234 lines cut ...]
[  4%] Built target getopt
[  3%] Built target doxygen_docs
Scanning dependencies of target uhd
[  4%] Building CXX object lib/CMakeFiles/uhd.dir/types/sensors.cpp.o
[  4%] Building CXX object lib/CMakeFiles/uhd.dir/types/sid.cpp.o
[  5%] Building CXX object lib/CMakeFiles/uhd.dir/types/serial.cpp.o
[  5%] Building CXX object lib/CMakeFiles/uhd.dir/types/time_spec.cpp.o
[  5%] Building CXX object lib/CMakeFiles/uhd.dir/types/tune.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
lib/CMakeFiles/uhd.dir/build.make:342: recipe for target 'lib/CMakeFiles/uhd.dir/types/sid.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd.dir/types/sid.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
lib/CMakeFiles/uhd.dir/build.make:318: recipe for target 'lib/CMakeFiles/uhd.dir/types/serial.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd.dir/types/serial.cpp.o] Error 4
CMakeFiles/Makefile2:117: recipe for target 'lib/CMakeFiles/uhd.dir/all' failed
make[1]: *** [lib/CMakeFiles/uhd.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

dkozel, 2017-10-30:
Hello @pikumar 

Errors like this are usually caused by GCC running out of memory. How much memory does your VM have? Are you using a parallel build?

Can you please try increasing your available RAM and try again? If you encounter the error again then checking if the build is single threaded may help.

pikumar, 2017-10-30:
Hi @dkozel

Thanks for the quick response. Indeed that was the issue. Please close this ticket.

dkozel, 2017-10-30:
@pikumar I'm glad that solved your issue. If you have any questions in the future the USRP-users mailing list is a great place to ask them.
http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com

closed 2017-10-30 by dkozel