EttusResearch/uhd #137: UHD underflows under saturation

FarshadKoohifar, 2017-10-31:
Hi,
I am putting UHD tx streamer under saturation load, yet I still get U indicators. 
Here is the relevant part of my code to reproduce it:

```
txmd.start_of_burst = true;
while (1){
	try{
		tx_stream->send(txTimeDomainSymbols,(size_t)(totalGeneratedSamples), txmd,0.2);
		txmd.start_of_burst = false;
		txmd.has_time_spec = false;
		txmd.end_of_burst = false;
	}catch(std::runtime_error& e) {}
}
```

My sample rate is 390625. My CPU is Core i7 6700k. I am connected over regular ethernet cable to X310 USRP. My UHD is UHD_003.010.002.000-3-g122bfae1. USRP image is not custom, with the following relevant information:

```
 FW Version: 5.1
 FPGA Version: 33.0
 RFNoC capable: Yes
```
How can I achieve the GnuRadio level of throughput without U? Should I use a different version of API?

Best regards,
Farshad Koohifar

creinke-sandia, 2017-11-29:
I gave 3.10.2 (and 3.10.1) a try, and I also get "U"s even though the output signal show no signs of underflow.  It doesn't happen with 3.9.7.

I am also using a X310 over Ethernet.

michael-west, 2018-02-16:
@FarshadKoohifar @creinke-sandia Thank you for the information.  This is a known issue on UHD 3.10.0.0 and later.  We are working to resolve it.  The simple workaround is to add the device argument "send_frame_size=4124" (or "send_frame_size=1052" if the Ethernet MTU is less than 4124) to the UHD USRP Sink, UHD USRP Source, and/or Device3 blocks.

michael-west, 2018-02-16:
See also #128

michael-west, 2018-03-27:
This issue is already being tracked internally and is being fixed.  Closing this issue to avoid tracking in multiple places.  Please use the workaround for now and watch future release notes for a fix.

closed 2018-03-27 by michael-west