EttusResearch/uhd #139: Can I install uhd on 32-bit Ubuntu ? Illegal instruction.

TriLoo, 2017-11-03:
Hi there, 
   I want install uhd  @'release_003_010_001_001 ' on my 32-bit 14.04 Ubuntu, however, I got a "Illegal Instruction" when i used 'from gnuradio import uhd'.

  Anyone had seen this problem and any advise? Thanksin advance.

dkozel, 2017-11-03:
Hello @TriLoo 

How did you install UHD and GNU Radio? Do the UHD examples run? Does GNU Radio work on it's own?

We have the following information about building and installing UHD and GNU Radio together which may be useful.
https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux

mbr0wn, 2017-11-03:
Are you on ARM by any chance? Or x86?

TriLoo, 2017-11-07:
@mbr0wn Thanks for reply, actually, I am installing it on my Jetson TX1 (ARM), ubuntu-14.04 32-bits. It works good on my ubuntu-16.04 64-bits (also TX1).

TriLoo, 2017-11-07:
@dkozel I installed all the dependencies listed on that page, but not include 'libqwtplot3d-qt4-dev', because there is no this candicate when I ran 'sudo apt-get install libqwtplot3d-qt4-dev'.

Then I installed the UHD using '**release_003_010_001_001** '

Then I installed the GNU Radio using '**3.7.12git**'

And, I installed a component individually called '**gr-usrpTRX**'.

Result was that the gnuradio-companion can run, however, when I ran the examples under 'gr-usrpTRX/apps',  I got the 'Illegal instruction'. Finally, I found that when I run 'from gnuradio import uhd ', 'Illegal instruction' pops out.

Thanks for reply ~

mbr0wn, 2017-11-07:
Please take this discussion to usrp-users, or even discuss-gnuradio. What's happening is probably one of the following:
- The toolchain is not being applied, and a cross compile is actually a native compile (i.e., trying to run x68 code on ARM)
- The toolchain is creating instructions not supported by the TX1 (e.g., wrong FPU instruction set). 

This is most likely not a bug in UHD (and if it is, we we reopen this issue).

closed 2017-11-07 by mbr0wn