EttusResearch/uhd #140: Python API not Enabled

ezze5, 2017-11-03:
Hi, 
So I need to install the Python API and I tried by installing this, however after running the cmake file the Python API is not there and I can't include it in my project, any idea why?  thank you.

######################################################
-- # UHD enabled components                              
-- ######################################################
--   * LibUHD
--   * LibUHD - C API
--   * Examples
--   * Utils
--   * Tests
--   * USB
--   * B100
--   * B200
--   * USRP1
--   * USRP2
--   * X300
--   * N230
--   * OctoClock
--   * Manual
--   * API/Doxygen
--   * Man Pages
-- 
-- ######################################################
-- # UHD disabled components                             
-- ######################################################
--   * GPSD
--   * E100
--   * E300

mbr0wn, 2017-11-03:
Can you confirm you did install from the `python-api` branch?

ezze5, 2017-11-06:
Hello mbr0wn, 
I cloned from git https://github.com/EttusResearch/uhd.git (this project) and then ran the cmake file that is under host folder since it was the one that seemed correct. thank you

dkozel, 2017-11-06:
Hello @ezze5 

Since you have cloned the git repository you'll need to checkout the python-api branch of UHD. Then rerun cmake and build the project.

ezze5, 2017-11-06:
Hello @dkozel thank you for your answer, but how exactly do I do this? Would it be in the cloned repository or somewhere else?
Is it modifying the cmake file? I cannot find any file or branch named as python API
 Maybe it is a simple question but I am really new in the topic.
Thank you

dkozel, 2017-11-06:
@ezze5 Without knowing what tool you used to clone the repository I can't point to a specific guide. There are many resources online.

If you used the command line then this page has some useful resources.
https://www.digitalocean.com/community/tutorials/how-to-use-git-branches

Github has the Try Git tutorial which is also a good place to become more familiar with Git.
https://try.github.io/

From within the repository directory run:
`git checkout python-api`

ezze5, 2017-11-06:
@dkozel Thank you very much.
I cloned it in the terminal with `git clone https://github.com/EttusResearch/uhd.git` and ran the cmakelist file from the host folder and ran `make` afterwards.
I decided to delete everything and clone it again as I changed some files trying to enable the python API. 
Now after cloning it again and checking out the python API a folder named python appeared, inside I can see a cmake file, is it the one I have to run? or shall I run the one that is at host? Thanks a lot and sorry for the silly question.

ezze5, 2017-11-06:
So, now I checked out python-api branch ran the host cmake file and used the command `make`, how ever when running the example pyuhd_rx_to_file.py I get an error at import uhd: ImportError: No module named uhd, am I missing a step?

mbr0wn, 2017-11-06:
Tagging as support issue and closing. @ezze5, please take the discussion to the mailing list where we are better set up to help you. This does not seem to be a bug in UHD.

closed 2017-11-06 by mbr0wn